FydeOS/chromium_os-raspberry_pi #2: Raspberry Official 7" Touchscreen support

rideorgtfo, 2018-11-22:
Touchscreen always black on [R70](https://github.com/FydeOS/chromium_os_for_raspberry_pi/tree/r70) build. I try plug and unplug hdmi, but it doesnot help.
But on [R68](https://github.com/FydeOS/chromium_os_for_raspberry_pi/tree/r68) it start itself, out-of-box, without hdmi manipulation.
How can I start it on R70?

puccaso, 2018-11-28:
i would also love a resolution for this!
same issue here

nmcain, 2019-01-15:
Perhaps we could merge the drivers for the screen into chrome os 70?

yunshansimon, 2019-01-18:
Can you provide `/var/log/message`? Chromium OS for rpi3 is based on RPI official kernel with official configure. You can compile the kernel yourself and copy it to sd card.

mac-user669, 2019-06-25:
I haven't installed FydeOS yet, but I have an Rpi 3b+ and the official 7 inch touchscreen. Has this issue been resolved? I can test, if need be

puccaso, 2019-08-27:
> Can you provide `/var/log/message`? Chromium OS for rpi3 is based on RPI official kernel with official configure. You can compile the kernel yourself and copy it to sd card.

Interesting.. Could we try the #gentoo kernel? Any blobs we should know about?

fydeos-alpha, 2019-11-21:
Bulk closure of dated issues

closed 2019-11-21 by fydeos-alpha

MarcKarasek, 2020-01-14:
Has this been resolved?

ankorez, 2020-02-29:
hello same issue here i have rollback to r68 :(