FydeOS/chromium_os-raspberry_pi #7: dev-libs/openssl fails

nblandfo, 2019-01-09:
Following the instructions in the README.txt, the following package is failing dev-libs/openssl-1.0.2p-r1. Log file is attached

[dev-libs:openssl-1.0.2p-r1:20190109-174102.log](https://github.com/FydeOS/chromium_os_for_raspberry_pi/files/2742310/dev-libs.openssl-1.0.2p-r1.20190109-174102.log)

yunshansimon, 2019-01-12:
The issue is caused by a file for testing. There is a workaround:
copy the patch file to `(cros)~/trunk/src/third_party/chromiumos-overlay/dev-libs/openssl/files` ,
insert `epatch "${FILESDIR}/remove_bad_dtls_test.patch"` into function `src_prepare()` in `openssl-1.0.2p.ebuild`  .

remove_bad_dtls_test.patch:

```
--- a/test/Makefile	2019-01-12 23:37:42.775725999 +0800
+++ b/test/Makefile	2019-01-12 23:45:51.138795233 +0800
@@ -70,7 +70,7 @@
 CONSTTIMETEST=  constant_time_test
 VERIFYEXTRATEST=	verify_extra_test
 CLIENTHELLOTEST=	clienthellotest
-BADDTLSTEST=	bad_dtls_test
+#BADDTLSTEST=	bad_dtls_test
 SSLV2CONFTEST = 	sslv2conftest
 DTLSTEST =	dtlstest
 FATALERRTEST =	fatalerrtest
@@ -88,7 +88,7 @@
 	$(ASN1TEST)$(EXE_EXT) $(V3NAMETEST)$(EXE_EXT) $(HEARTBEATTEST)$(EXE_EXT) \
 	$(CONSTTIMETEST)$(EXE_EXT) $(VERIFYEXTRATEST)$(EXE_EXT) \
 	$(CLIENTHELLOTEST)$(EXE_EXT) $(SSLV2CONFTEST)$(EXE_EXT) $(DTLSTEST)$(EXE_EXT) \
-	$(BADDTLSTEST)$(EXE_EXT) $(FATALERRTEST)$(EXE_EXT) $(X509TIMETEST)$(EXE_EXT)
+	$(FATALERRTEST)$(EXE_EXT) $(X509TIMETEST)$(EXE_EXT)
 
 # $(METHTEST)$(EXE_EXT)
 
@@ -103,7 +103,7 @@
 	$(EVPTEST).o $(EVPEXTRATEST).o $(IGETEST).o $(JPAKETEST).o $(ASN1TEST).o $(V3NAMETEST).o \
 	$(HEARTBEATTEST).o $(CONSTTIMETEST).o $(VERIFYEXTRATEST).o \
 	$(CLIENTHELLOTEST).o  $(SSLV2CONFTEST).o $(DTLSTEST).o ssltestlib.o \
-	$(BADDTLSTEST).o $(FATALERRTEST).o $(X509TIMETEST).o
+	$(FATALERRTEST).o $(X509TIMETEST).o
 
 SRC=	$(BNTEST).c $(ECTEST).c  $(ECDSATEST).c $(ECDHTEST).c $(IDEATEST).c \
 	$(MD2TEST).c  $(MD4TEST).c $(MD5TEST).c \
@@ -115,7 +115,7 @@
 	$(EVPTEST).c $(EVPEXTRATEST).c $(IGETEST).c $(JPAKETEST).c $(SRPTEST).c $(ASN1TEST).c \
 	$(V3NAMETEST).c $(HEARTBEATTEST).c $(CONSTTIMETEST).c $(VERIFYEXTRATEST).c \
 	$(CLIENTHELLOTEST).c  $(SSLV2CONFTEST).c $(DTLSTEST).c ssltestlib.c \
-	$(BADDTLSTEST).c $(FATALERRTEST).c $(X509TIMETEST).c
+	$(FATALERRTEST).c $(X509TIMETEST).c
 
 EXHEADER= 
 HEADER=	testutil.h ssltestlib.h $(EXHEADER)
@@ -374,9 +374,9 @@
 	@echo $(START) $@
 	../util/shlib_wrap.sh ./$(CLIENTHELLOTEST)
 
-test_bad_dtls: $(BADDTLSTEST)$(EXE_EXT)
-	@echo $(START) $@
-	../util/shlib_wrap.sh ./$(BADDTLSTEST)
+#test_bad_dtls: $(BADDTLSTEST)$(EXE_EXT)
+#	@echo $(START) $@
+#	../util/shlib_wrap.sh ./$(BADDTLSTEST)
 
 test_fatalerr: $(FATALERRTEST)$(EXE_EXT)
 	@echo $(START) $@
@@ -571,8 +571,8 @@
 $(CLIENTHELLOTEST)$(EXE_EXT): $(CLIENTHELLOTEST).o
 	+@target=$(CLIENTHELLOTEST) $(BUILD_CMD)
 
-$(BADDTLSTEST)$(EXE_EXT): $(BADDTLSTEST).o
-	+@target=$(BADDTLSTEST) $(BUILD_CMD)
+#$(BADDTLSTEST)$(EXE_EXT): $(BADDTLSTEST).o
+#	+@target=$(BADDTLSTEST) $(BUILD_CMD)
 
 $(FATALERRTEST)$(EXE_EXT): $(FATALERRTEST).o ssltestlib.o $(DLIBSSL) $(DLIBCRYPTO)
 	+@target=$(FATALERRTEST); exobj=ssltestlib.o; $(BUILD_CMD)
```

nblandfo, 2019-01-14:
The patch is failing.
`* ERROR: dev-libs/openssl-1.0.2p-r1::chromiumos failed (prepare phase):
 *   Failed Patch: remove_bad_dtls_test.patch!
 * 
 * Call stack:
 *     ebuild.sh, line  133:  Called src_prepare
 *   environment, line 3218:  Called epatch '/mnt/host/source/src/third_party/chromiumos-overlay/dev-libs/openssl/files/remove_bad_dtls_test.patch'
 *   environment, line 1121:  Called die
 * The specific snippet of code:
 *               die "Failed Patch: ${patchname}!";`

[dev-libs:openssl-1.0.2p-r1:20190114-174327.log](https://github.com/FydeOS/chromium_os_for_raspberry_pi/files/2756498/dev-libs.openssl-1.0.2p-r1.20190114-174327.log)

Thanks

yunshansimon, 2019-01-16:
openssl-1.0.2p.ebuild

```
 if ! use vanilla ; then
        epatch "${WORKDIR}"/patch/*.patch

        # Chromium OS patches.
        epatch "${FILESDIR}"/${PN}-1.0.2-blacklist-by-sha1.patch
        epatch "${FILESDIR}/remove_bad_dtls_test.patch"
    fi

    epatch_user
```
Insert the epatch command at the last of  `src_prepare` patch process.

fydeos-alpha, 2019-11-21:
bulk closure of dated issues

closed 2019-11-21 by fydeos-alpha