FydeOS/chromium_os-raspberry_pi #9: sudo default password?

HansLanger, 2019-01-15:
Hi, 
What is the default SUDO password?

I came through this link: https://www.raspberrypi.org/forums/viewtopic.php?t=227281
I downloaded the bin, with etcher I wrote the bin into an SD, I started my raspi3b+ with the SD, and its immediately started chrome os, great work! thank you!.
Then,I open the terminal, I shell, and when I write "sudo command" and it ask me for a password. I can't do anything that I need without sudo privileges.

Thanks.

HansLanger, 2019-01-15:
Nevermind, the password was right there. I let this post in case someone has the same issue:
password: test0000

xNUTs, 2019-10-27:
Use Terminal sudo -s 
Enter PASSWD to create a password.

closed 2020-02-14 by fydeos-alpha