FydeOS/chromium_os-raspberry_pi #10: Only 2gb of sd card space is available, even on 32 gb card.

nmcain, 2019-01-16:
In GParted, it shows tons of empty space. Can we fix this, please.

yunshansimon, 2019-01-16:
[expand.sh.txt](https://github.com/FydeOS/chromium_os_for_raspberry_pi/files/2762885/expand.sh.txt)

1.save the script to `expand.sh`
2.change mod `chmod 755 expand.sh`
3.run as root `sudo expand.sh /dev/sda` or /dev/mmcblk0 
4.restart os.

yunshansimon, 2019-01-16:
It doesn't belong vanilla Chromium OS. It is from FydeOS.

nmcain, 2019-01-16:
when I run the script, it shows this.
`chronos@(none) ~/Downloads $ chmod 755 expand.sh
chronos@(none) ~/Downloads $ sudo expand.sh /dev/sda
sudo: expand.sh: command not found
chronos@(none) ~/Downloads $ ls
expand.sh  IMG_20181229_175907.jpg
chronos@(none) ~/Downloads $ expand.sh /dev/sda
bash: expand.sh: command not found
chronos@(none) ~/Downloads $ ./expand.sh /dev/sda
bash: ./expand.sh: Permission denied
chronos@(none) ~/Downloads $ sudo ./expand.sh /dev/sda
sudo: unable to execute ./expand.sh: Permission denied
chronos@(none) ~/Downloads $ chmod +x expand.sh
chronos@(none) ~/Downloads $ sudo expand.sh /dev/sda
sudo: expand.sh: command not found
chronos@(none) ~/Downloads $ ./expand.sh
bash: ./expand.sh: Permission denied
chronos@(none) ~/Downloads $ sudo ./expand.sh
sudo: unable to execute ./expand.sh: Permission denied
chronos@(none) ~/Downloads $ `

yunshansimon, 2019-01-17:
The execute permission of ~/Downloads is forbiden. You need copy it to `/usr/local/bin` or `/usr/local/sbin`.

nmcain, 2019-01-17:
`(none) /usr/local # sudo ./expand.sh /dev/sda
ERROR: Can't open /dev/sda: No such file or directory
ERROR: Can't open /dev/sda: No such file or directory
ERROR: Can't open /dev/sda: No such file or directory
ERROR: Can't open /dev/sda: No such file or directory
partx: stat of /dev/sda failed: No such file or directory
resize2fs 1.44.1 (24-Mar-2018)
open: No such file or directory while opening /dev/sda1
(none) /usr/local # sudo ./expand.sh /dev/mmcblk0
ERROR: Entry outside of valid region
ERROR: -i 1 -s 59306940 
resize2fs 1.44.1 (24-Mar-2018)
The filesystem is already 524288 (4k) blocks long.  Nothing to do!

(none) /usr/local # `

jslby, 2019-01-19:
@yunshansimon 
can you help me pls.
`sudo ./expand.sh /dev/mmcblk0                                                             ~/Downloads
ERROR: Entry outside of valid region
ERROR: -i 1 -s 60245948
resize2fs 1.44.5 (15-Dec-2018)
Filesystem at /dev/mmcblk0p1 is mounted on /media/STATE; on-line resizing required
old_desc_blocks = 1, new_desc_blocks = 1
The filesystem on /dev/mmcblk0p1 is now 524301 (4k) blocks long.
`

yunshansimon, 2019-01-29:
Try the latest release

blairbo, 2019-02-23:
This worked perfectly for me.

MuTong233, 2019-05-03:
Try the latest build and login as a normal Google account.It should be expanded automatically.

PrinceAlvar, 2019-05-13:
hello im new to Chrome OS and i currently have this problem, how do i get to the Chrome OS terminal and what command do i enter into it so that it reads the full 64gb SD card i have?

liberodark, 2019-12-11:
please add this script into the images would be better.

PrinceAlvar, 2020-02-11:
> please add this script into the images would be better.
> in /usr/bin/ or run that in auto?

can you please walk me through on how to do this over call or discord?
#6199409474
Disc #0804 Prince_Alvar

closed 2020-02-14 by fydeos-alpha

yetipi3, 2020-07-23:
#80 i dont understand (im new to this)

fydeos-alpha, 2020-08-28:
- First, bring up a shell prompt (ctrl + alt + F2 for tty shell; or ctrl + alt + t for crosh shell then enter `shell` for bash), login with `chronos` / `test0000`

- Execute the `lsblk` command and verify your stateful partition label name, it's likely to be `mmcblk0p1`. Make sure it's listed in the output.

- Execute the expansion script:
   ```
   sudo /usr/sbin/expand-partition.sh --dst /dev/mmcblk0p1
   ```
   Once the script finish running without error, your stateful partition should be expanded to occupy the rest of your SD card free space.

vedranius, 2021-01-19:
> * First, bring up a shell prompt (ctrl + alt + F2 for tty shell; or ctrl + alt + t for crosh shell then enter `shell` for bash), login with `chronos` / `test0000`
> * Execute the `lsblk` command and [...]

Thanks! This worked great! :)