FydeOS/chromium_os-raspberry_pi #11: Raspberry Pi 3B+ GPU setting

briantham, 2019-01-18:
Hi,

Thanks for the great work. I follow the instruction to install, the chrome browser is working fine. But when I use it to watch some .ts streaming video, it start to lag and choppy, and it finally stop rendering the video. So I start to realize may be I need to tune up the GPU memory for my raspi3b+, I tried to modify the config.txt file and change gpu_mem=16 (default ?) to gpu_mem=192, but it still the same, Now I have question,

Does this option under config.txt is working after kernel load up ? and Should I buy the MPEG2 & VC-1 license and add it under config.txt, will it take effect for better performance ?

Seems the chrome://gpu show up using hardware acceleration, but I want to confirm the config.txt is working under FydeOS before I buy the license. 

Thanks.

yunshansimon, 2019-01-18:
I have no idea about the license about MPEG2.  But the next version will be updated to Chromium OS R72 with some tunes for performance. Let's see what will happen. BTW, it is pure Chromium OS, not FydeOS.

nmcain, 2019-01-19:
How do you modify config.txt?

yunshansimon, 2019-01-29:
According to `https://www.raspberrypi.org/documentation/configuration/config-txt/`

NamedJohnny, 2019-02-09:
Do we know if Chromium OS use the config.txt file? I tried to modify it but it didn't work. After I burn the .img to the Micro SD card and modify the file I got a : Write Protection Error. But burning Raspbian or other OS .img, modifying the config.txt file work.

yunshansimon, 2019-04-16:
In fact, Rpi 3B(+) plays the video with software decoder. There is no compatible interface for chromium to visit the hardware decoder of VC4. We need the hackers or Broadcom official to provide compatible driver fro chromium os.

briantham, 2019-04-16:
according to https://www.raspberrypi.org/documentation/configuration/config-txt/, Rpi 3B(+) will use the config.txt. Also, the Chrome browser able to use the HW decoder. I think it should be the OS level to communicate with HW decoder, at lease I found the config.txt able to use by Raspberry OS.

closed 2020-12-12 by fydeos-alpha