FydeOS/chromium_os-raspberry_pi #14: Distorted Screen on Boot

midhunvnadh, 2019-01-23:
I am Getting a Distorted Screen on Boot. I am guessing that there are clickable items / Buttons Below the Distorted Screen. It is a display issue ?. Do I need to edit in the config.txt file ?. Please Help

midhunvnadh, 2019-01-28:
AOC E1670SWU 15.6-inch LED Monitor
Connection via HDMI to monitor, Pi power supply.
It's the Pi 3 B+ Itself.

How can I get to the dmesg log ?
Photo Uploaded.
[image]

yunshansimon, 2019-01-29:
Does your monitor work with official Rasbian?

midhunvnadh, 2019-01-29:
Yes. It works !. Fantastically !.

midhunvnadh, 2019-02-01:
Guys !. Help Please

yunshansimon, 2019-02-02:
Try comment “hdmi_drive=2” in config.txt at /dev/sda12

tresni, 2019-02-03:
I'm having the same issue, commented out `hdmi_drive=2` in config.txt and tried again, same results.  My TV is a Vizio E320-B0.  Works fine with Rasbian (my normal config.txt for Raspbian
 only has `dtparam=audio=on` and nothing else.)  I'm not sure what's safe to comment out in the config.txt to continue testing.

[image]

midhunvnadh, 2019-02-03:
No luck to me too :(

akavkl, 2019-02-16:
Dev/sda12 not found on r72

benklop, 2019-05-18:
what's strange is when this happens for my brother's install, the mouse cursor is not distorted

raynmune, 2019-07-30:
Same issue here. Monitor distorted like @tresni and mouse cursor not distorted, can play around clicking and distort changes

sidifal, 2019-08-03:
any body have solution!!

sincoew, 2019-12-03:
same issue +1

tresni, 2020-01-21:
Just as a note, I plugged the rPI into a standard monitor and it worked.  I ended up giving up on trying to make this work on the TV.

LCCR193, 2020-08-08:
> Just as a note, I plugged the rPI into a standard monitor and it worked. I ended up giving up on trying to make this work on the TV.

Well, but on the monitor I have it is distoted, (resolution 1366x768@60hz, 18.5")

muratcesmecioglu, 2020-10-30:
same here. please help.

16110267, 2020-12-09:
hi i have this issue too, anyone know how to fix this issue?

muratcesmecioglu, 2020-12-09:
> hi i have this issue too, anyone know how to fix this issue?

I just tried another (and worse) monitor with default installation and it worked. :)

closed 2020-12-12 by fydeos-alpha