FydeOS/chromium_os-raspberry_pi #15: RPI3 B+ Screen issue

PAPADON69, 2019-01-26:
when i boot it up the screen is not clear. i mean to say is that the the screen is not displaying

fydeos-alpha, 2019-01-26:
you'd have to supply some more information to let this issue be any meaningful.

e.g.: screen type, make and model; connection setup; dmesg log; photo of how 'not clear' or 'not displaying'

sincoew, 2019-12-03:
hi,
I have same issue,
my screen type:
ZHIXIANDA H1208 8 inch TFT LCD 1024x768 Monitor with VGA HDMI AV BNC USB Sperkers for PC CCTV Security Camera

Model Number | H1208
-- | --
Screen Ratio | 4:3 TFT LCD panel
Resolution | 1024*768
Contrast | 800:1
Brightness | 350cd/m2
Response Time | 25 (Typ.)(Tr+Td) (ms)
Viewing Angle | Horizontal(left/right): 170°(85°/85°)  Vertical(up/down):    170°(85°/85°)
Backlight | LED backlight
Color system | PAL/ NTSC
Power | DC12V 1A
Interface | HDMI/BNC/VGA/AV/USB
Support Languages | English,Traditional Chinese,Simplified  Chinese,Francais,Italiano,Deutsch,  Espanol,Russian
Plug | AU,EU,USA,UK plug available
Monitor Size | 197*161*35mm
Active Area/Display Area | 164*123mm
Delivery Time | Sample:1~3 working days.
10~100pcs:2~5 working days.
More than 100pcs :5~15 working days.