FydeOS/chromium_os-raspberry_pi #16: Latest build (r72) still does not work with 7in touch screen

nmcain, 2019-01-29:
It works over hdmi, but the touch screen is black.

akavkl, 2019-02-16:
In r72 HDMI driver 2 forces to HDMI hence dsi doesn't work

yunshansimon, 2019-02-19:
Any flags in _config.txt_ like `hdmi_drive=2` are editable. You can modify them regard as your hardware.  `dtoverlay=vc4-kms-v3d,cma-128` is only necessary for Chromium OS.

nmcain, 2019-02-19:
How do I modify config.txt inside of chromium os?

akavkl, 2019-02-19:
Write protected hence couldn't change it

akavkl, 2019-02-19:
Only possible way now is using ssh and replacing it with modified one

yunshansimon, 2019-02-20:
Does it work? If it does, please share your knowledge about it.

nmcain, 2019-02-22:
How do I get the touch screen working??

Gordon1974, 2019-03-19:
Anyone found a resolution to this yet?

MuTong233, 2019-05-03:
Yeah,I am using a 7inch touch screen,but my screen resolution is 1024x600.It works very well and everything is OK.At first,My screen is blank too,I solved it by a 3rd party power adapter.Connect the power/touch cable to a power adapter such as smartphone charger and after logged in,put the cable back.I think it's because the HDMI detection can't work very well.

MarcKarasek, 2020-01-14:
Any resolution to this?  Would like to get this working on a7" touch screen.   

Did you have a hdmi connected and then switched over to touch?

yunshansimon, 2020-01-20:
At last, we got an official 7inch display. The magic to fix this issue is changing `dtoverlay=vc4-kms-v3d,cma-128` to `dtoverlay=vc4-fkms-v3d,cma-128` in `config.txt` .  Good luck!

xmapst, 2020-02-01:
Does it support output to the official 7-inch touch display of raspberrypi through the display dsi interface?

xmapst, 2020-02-01:
Which version? What needs to be configured?

fydeos-alpha, 2020-02-14:
@xmapst 
This issue was about the compatibility of the official 7-inch display with Raspberry Pi 3B / 3B+ releases. All releases were affected. 

Releases on Raspberry Pi 4B is not affected.

The solution has been provided above, by adding some extra configs to `config.txt` under partition 12.

A couple of methods to modify this `config.txt`:
 - Boot target Chromium OS on a Pi, but with a different display setup so that you can see UI. Enter tty shell by hitting `ctrl` + `alt` + `F2`, log in with username `chronos`. Issue the following command to:
      - Gain sudo access and read/write permission
      - Create a mount point
      - Mount partition 12
      - Do your modification
      - Save, un-mount, reboot
      ```
      sudo su
      mount -o remount rw /
      mkdir /tmp/p12
      mount /dev/sda12 /tmp/p12
      
      vim /tmp/p12/config.txt    
      #do your changes by changing dtoverlay=vc4-fkms-v3d,cma-128
      
      umount /tmp/p12
      reboot
      ```

 - Or you can directly modify chromiumos_test_image_<version_string>.img on a Linux machine by mounting its partition 12 and do your modifications:

```
# uncompress your image file first
xz -d  chromiumos_test_image_<version_string>.img.xz

# add its partition mappings to your host
sudo kpartx -a chromiumos_test_image_<version_string>.img

# list its partition mappings, take note of the loop device it has created
sudo kpartx -l chromiumos_test_image_<version_string>.img

# We will use loop0 as an example, create a mount point, mount it
sudo mkdir /mnt/crosp12
sudo mount -t ext2 /dev/mapper/loop0p12 -o ro /mnt/crosp12

# do your change, by changing dtoverlay=vc4-fkms-v3d,cma-128
sudo vim /mnt/crosp12/config.txt

# Once changes are made and saved, delete the mappings
sudo kpartx -d chromiumos_test_image_<version_string>.img
```

closed 2020-02-14 by fydeos-alpha

gchino, 2020-02-15:
> mount /dev/sda12 /tmp/p12
At this line, i received an error stating that special device /dev/sda12 does not exist
how can i rectify this?

ethanmckellar, 2020-06-19:
After using 

> @xmapst
> This issue was about the compatibility of the official 7-inch display with Raspberry Pi 3B / 3B+ releases. All releases were affected.
> 
> Releases on Raspberry Pi 4B is not affected.
> 
> The solution has been provided above, by adding some extra configs to `config.txt` under partition 12.
> 
> A couple of methods to modify this `config.txt`:
> 
> * Boot target Chromium OS on a Pi, but with a different display setup so that you can see UI. Enter tty shell by hitting `ctrl` + `alt` + `F2`, log in with username `chronos`. Issue the following command to:
>   
>   * Gain sudo access and read/write permission
>   * Create a mount point
>   * Mount partition 12: it should be mmcblk0p12, check `lsblk` if it is
>   * Do your modification
>   * Save, un-mount, reboot
>   
>   ```
>   sudo su
>   mount -o remount rw /
>   mkdir /tmp/p12
>   mount /dev/mmcblk0p12 /tmp/p12
>   
>   vim /tmp/p12/config.txt    
>   #do your changes by changing dtoverlay=vc4-fkms-v3d,cma-128
>   
>   umount /tmp/p12
>   reboot
>   ```
> * Or you can directly modify chromiumos_test_image_<version_string>.img on a Linux machine by mounting its partition 12 and do your modifications:
> 
> ```
> # uncompress your image file first
> xz -d  chromiumos_test_image_<version_string>.img.xz
> 
> # add its partition mappings to your host
> sudo kpartx -a chromiumos_test_image_<version_string>.img
> 
> # list its partition mappings, take note of the loop device it has created
> sudo kpartx -l chromiumos_test_image_<version_string>.img
> 
> # We will use loop0 as an example, create a mount point, mount it
> sudo mkdir /mnt/crosp12
> sudo mount -t ext2 /dev/mapper/loop0p12 -o ro /mnt/crosp12
> 
> # do your change, by changing dtoverlay=vc4-fkms-v3d,cma-128
> sudo vim /mnt/crosp12/config.txt
> 
> # Once changes are made and saved, delete the mappings
> sudo kpartx -d chromiumos_test_image_<version_string>.img
> ```
 the display shows up on the screen but touch controls do not work what so ever

zan73, 2020-08-04:
The line:
`sudo mount -t ext2 /dev/mapper/loop0p12 -o ro /mnt/crosp12`
Should be:
`sudo mount -t msdos /dev/mapper/loop0p12 -o rw /mnt/crosp12`

nico151999, 2020-09-23:
Is there a list of supported touch screens? Would prefer a screen around 10".

nmcain, 2020-09-24:
From my testing, you want something that does not require a Goodix or Silead touch driver.