FydeOS/chromium_os-raspberry_pi #17: No Sound over HDMI

tomasmark79, 2019-02-02:
Hello.

I was trying this build on my raspberry 3b+ and none sound over HDMI cable to my TV Samsung.

I tried mount partition with the config.txt and chanchged the hdmi%driver parametr to comment, but no result. Only Jack audio is working, but HDMI not.

Thank you for any answer and help.

yunshansimon, 2019-02-03:
For some reasons, the sound driver of Rpi3 isn't compatible with CRAS. The default output mode is "auto" which can't output to either HDMI or headphone. So, we wrote a configuration file to force to output channel to the headphone jack. To change it to HDMI, you need to edit the file at `/etc/init/force_audio_output_to_headphones.conf`. There also contains a message to how to switch between two channels.

nledenyi, 2019-02-03:
Can you please point me to a guide which describes how to access a shell within Chromium OS to edit that file? I simply just downloaded the image to an sd card and booted it with RPi3+, didn't build it from source. Is not possible in that case?

yunshansimon, 2019-02-04:
1. login
2. Ctrl-Alt-T open a shell tab
3. type `shell` to run bash
4. Modify the file
```
$ sudo mount -o remount,rw /
$ sudo vi /etc/init/force_audio_output_to_headphones.conf
```

nledenyi, 2019-02-04:
Thanks, worked like a charm. As someone starting from scratch, I found this, together with other two issues (issues/9 and issues/10) really useful to get started - would be nice to persist these somewhere for others before they disappear from the issues.

closed 2019-02-07 by yunshansimon

MartGrat, 2019-05-02:
Hello I have the same Issue with my Pipo x8 (z3637f intel atom)  I got no sound hdmi.  I tried to find the file you mentioned to edit it but I could’nt find it in the iso file

sgtram, 2020-03-26:
Hi, whatever I set 2 or 0 the sound always headphones. No HDMI audio. Any idea?

adsports, 2020-12-13:
> 1. login
> 2. Ctrl-Alt-T open a shell tab
> 3. type `shell` to run bash
> 4. Modify the file
> 
> ```
> $ sudo mount -o remount,rw /
> $ sudo vi /etc/init/force_audio_output_to_headphones.conf
> ```

hello
I try to enter with sudo but it tells me, ¨we trust have ..... ¨ and asks for my password.

How can I give write permission?