FydeOS/chromium_os-raspberry_pi #18: Constant Crashing on newest build

Hasplit, 2019-02-13:
When I start up the device it allows me to login before crashing and restarting the system. I'm going to try the previous build soon as it may just be my device or sd card.

StuartIanNaylor, 2019-04-30:
https://github.com/FydeOS/chromium_os_for_raspberry_pi/releases/download/r72/chromiumos_image_r72.img.xz On a Pi3b seems to work well.

swapanshridhar, 2019-10-13:
> https://github.com/FydeOS/chromium_os_for_raspberry_pi/releases/download/r72/chromiumos_image_r72.img.xz On a Pi3b seems to work well.

Tried the r72 too. This too reboots as soon as i select some file.

yunshansimon, 2019-10-14:
Check your power supply, replace it with a stable one. It needs more power than official raspbian(initial the GPU at the booting).

fydeos-alpha, 2019-11-21:
bulk closure of dated issues

rslashpuzzleheadedact, 2023-04-27:
Your desktop pc is incompatible I used to have a hp desktop that was not compatible one laptop that works is 300$ its the lenovo ideapad 3. press f12 to boot menu and f2 for bios.

Damenly, 2023-04-27:
> Your desktop pc is incompatible I used to have a hp desktop that was not compatible one laptop that works is 300$ its the lenovo ideapad 3. press f12 to boot menu and f2 for bios.

Sorry. It’s an issue board of chromiumos rpi. I’m confused why you talked about PC?

closed 2023-04-27 by Damenly

Scriptyure, 2023-04-27:
> > Your desktop pc is incompatible I used to have a hp desktop that was not compatible one laptop that works is 300$ its the lenovo ideapad 3. press f12 to boot menu and f2 for bios.
> 
> Sorry. It’s an issue board of chromiumos rpi. I’m confused why you talked about PC?

Not to mention this has been necroed from 4 years ago