FydeOS/chromium_os-raspberry_pi #21: Widevine CDM, Netflix

blairbo, 2019-02-23:
Firstly, thank you for your work on this.

Yesterday my cloudready laptop stopped playing Netflix properly, so while I investigated that, I took the opportunity to check out alternative options and found that I could install chromium os on my RPi 3B+ using this project. Hooray! When it came up immediately after imaging the microSD card, I was thrilled. 

That thrill fizzled (womp womp) when I fired up Netflix, and got error code C7701-1003: We cannot find all the required components to play Netflix on this device. Please visit chrome://plugins, ensure that both the "Netflix" and the "Widevine Content Decryption Module" plug-ins are enabled and their "Always allowed" setting is checked. 

Now, while the message itself is targeted at Chrome-the-browser, however, it should not be that difficult to get this working on chromium.

Any help would be greatly appreciated.

fydeos-alpha, 2019-02-28:
Thank you for your interests in Chromium OS for Raspberry Pi.

You should know that widevine CDM plugin is proprietary to Google and it's only shipped with Chrome OS official build. The 'normal' build process for Chromium OS no longer includes widevine.

As the maintainer for Chromium OS for Raspberry Pi, we don't really have the urge to obtain commercial license for shipping Google's proprietary stuff. Apologies.

prodrigestivill, 2019-03-07:
It might be done by installing it unofficially like [this](https://neverware.zendesk.com/hc/en-us/community/posts/360021790753/comments/360004140934).

But using this script instead in a separate full GNU/Linux computer: [ruario/widevine-flash_armhf.sh](https://gist.github.com/ruario/19a28d98d29d34ec9b184c42e5f8bf29).
And then only extract the generated tarball as readwrite root user.

But this will be lost on each update.

kiai72, 2019-08-30:
seems not to work for me. the script/tar places the libwidevine.so to /opt/google/chrome.
is this the correct place ?
(Tested on the 72 image from https://github.com/FydeOS/chromium_os_for_raspberry_pi/releases)

Doing the same on raspbian works - but here you have to place it to /usr/lib

closed 2019-10-23 by fydeos-alpha