FydeOS/chromium_os-raspberry_pi #23: Only the device owner can change certain settings

Void-Frost, 2019-04-10:
In an attempt to use the Linux app/Google Play app features I tried to switch the device's channel, but it wont let me, even though my account is the device's owner account.

Void-Frost, 2019-04-10:
[image]

pcs3rd, 2019-04-10:
@FydeOS, could you provide a pre built image *without* device provisioning for people who aren't in China? It could be viewed as a security risk.

pcs3rd, 2019-04-10:
Actually, @Void-Frost , [This might work for us. depends.](https://www.ifixit.com/Answers/View/231737/How+can+I+reset+a+managed+chrome+book+back+to+factory)

Void-Frost, 2019-04-10:
> Actually, @Void-Frost , [This might work for us.](https://www.ifixit.com/Answers/View/231737/How+can+I+reset+a+managed+chrome+book+back+to+factory) Then do a powerwash if possible? I haven't played with it too much. But, how can there be two owners?

I don't know how there could be multiple owners but I have tried powerwashing, It says "Restart to powerwash, and then you will be asked to confirm" and I click okay, It restarts, and it gives the SAME message as before.

Void-Frost, 2019-04-10:
Also what is the power button on fyde?

Void-Frost, 2019-04-11:
@pcs3rd

fydeos-alpha, 2019-04-12:
@Void-Frost Please note that this is Chromium OS for Pi that we are discussing here, not FydeOS. Therefore no Linux, no Android (there will never be Play which is proprietary to Google)

@pcs3rd device-provisioning / management wasn't intentional at the first place, as mentioned in the README, the image was built with vanilla Chromium branches and default build flags. In terms of security risk, the image provided here is of a TEST image, it's INSECURE by nature. You should never use this in any production or similar environment.

Void-Frost, 2019-04-12:
@fydeos-alpha Why can't their be a way for the OWNER of the device to change the settings that he wants to? It's a test build anyways. Also, why not add Google play+Linux support?

Void-Frost, 2019-04-16:
Also, @fydeos-alpha, what is the power button on Raspberry Pi Chromium OS?

yunshansimon, 2019-04-16:
> @fydeos-alpha Why can't their be a way for the OWNER of the device to change the settings that he wants to? It's a test build anyways. Also, why not add Google play+Linux support?

The first question, there need a TPM-chip or a software-TPM emulator to enable this feature which doesn't exist on RPi. The second, there are no such supporting from google. You can ask Google, I want to know the answer too.

pcs3rd, 2019-04-28:
Sorry for the inactivity @Void-Frost I am not getting notifications or updates for this issue.
@fydeos-alpha, I do understand the concern with using this in production enviroments, but I use this on a home network for nothing valuable. (If my network were comprimised, the test build of chromeos would be less of concern than the pc I do taxes and whatnot on, but risk is still not voided). Also, on chromium os derevitives like cloudready, the feature (smart unlock) is available, it just does not work properly (probably because there is no hardware level tpm or authentication). Also @yunshansimon [Here is something like what you are talking about.](https://www.zymbit.com/securing-raspberry-pi/.)

Sources [here](https://neverware.zendesk.com/hc/en-us/community/posts/223158067-Feature-NOT-working-Unlock-your-Chromebook-with-your-Android-phone)
[and here](https://neverware.zendesk.com/hc/en-us/community/posts/360002205207-How-to-unlock-using-android-phone-)

closed 2019-10-23 by fydeos-alpha

Piflyer, 2019-11-11:
You could switch to canary by running `live_in_a_coal_mine` on crosh.