FydeOS/chromium_os-raspberry_pi #27: emerge: there are no ebuilds to satisfy "chromeos-base/chromeos-config-bsp" for /build/rpi3/.

vpelss, 2019-06-15:
I have been trying to compile my first ChromeOS these past 2 weeks. I believe I have got all my ducks in line as per the readme (and on the ChromeOS page).

But I get the following error on building the packages: ./build_packages --board=rpi3

I am still trying to discover what might be causing this, but I thought I might as well ask.

I am using Linux Mint 19.1

-------------------------

!!! CONFIG_PROTECT is empty for '/build/rpi3/'

These are the packages that would be merged, in order:

Calculating dependencies  . .... done!
emerge: there are no ebuilds to satisfy "chromeos-base/chromeos-config-bsp" for /build/rpi3/.
emerge: there are no ebuilds to satisfy "chromeos-base/chromeos-config-bsp" for /build/rpi3/.
(dependency required by "virtual/chromeos-config-bsp-1-r1::chromiumos" [ebuild])
(dependency required by "chromeos-base/chromeos-config-0.0.1-r31::chromiumos" [ebuild])
(dependency required by "chromeos-base/power_manager-0.0.2-r3014::chromiumos[unibuild]" [ebuild])
(dependency required by "virtual/target-chromium-os-1-r124::chromiumos[power_management]" [ebuild])
(dependency required by "virtual/target-os-1-r3::chromiumos" [ebuild])
(dependency required by "virtual/target-os" [argument])
ERROR   : emerge detected broken ebuilds. See error message above.

yunshansimon, 2019-12-11:
The issue is solved.

akshaychoudhari358, 2020-09-22:
Hi, 
I am getting same error while building the r83 branch code.
Please provide me some input how you have solve the issue.

Thank you.

closed 2020-12-12 by fydeos-alpha