FydeOS/chromium_os-raspberry_pi #28: Developer mode

ShalokShalom, 2019-06-17:
Hi 🤗 

Is there a method to enable the developer mode in order to install apps like this:

https://code.headmelted.com/

nmcain, 2019-06-18:
Developer mode is already enabled, you can try crouton, but performance isn't great.

ShalokShalom, 2019-06-18:
I tried to install https://chrome.google.com/webstore/detail/crouton-integration/gcpneefbbnfalgjniomfjknbcgkbijom while no installation button appears.

ShalokShalom, 2019-06-18:
What is with the official Linux support by Google: 

[image]

fydeos-alpha, 2019-10-23:
@ShalokShalom that's Linux subsystem that you were referring to, due to the hardware performance capability we don't have plans to include this.

closed 2019-10-23 by fydeos-alpha