FydeOS/chromium_os-raspberry_pi #30: Plugins

ShalokShalom, 2019-06-17:
How is there no mention at all about Chrome plugins?

I see zero on the homepage and here in the Readme. 

I guess it is currently unsupported?

nmcain, 2019-06-18:
extensions are supported unless they require a DRM, which google is legally unable to license.

ShalokShalom, 2019-06-18:
I have tried several ones, including uBlock Origin, which is an open source app.

closed 2019-10-23 by fydeos-alpha

ShalokShalom, 2019-11-06:
Nothing works and you close this issue?

Is this the quality control in this project?

fydeos-alpha, 2019-11-06:
My understanding to your original question was about "Were plugins supported in Chromium OS for Raspberry Pi", which was well answered.

If you wished to report things don't work on your setup, it'd be helpful to rephrase with focuses on the issue itself and providing more information / screenshot / logfile will always help. Spreading FUD won't.

ShalokShalom, 2019-11-06:
My second comment means that I tried them and it was not working. 

That there is no documentation about whatever they are working or not, is still its own issue. 

Pointing out a documentation issue is no FUD, it is part of the development.