FydeOS/chromium_os-raspberry_pi #34: Raspberry Pi 3b+ black screen after boot.

filipton12, 2019-08-28:
HI!
If i flash image to usb and i boot rpi i have a rainbow screen for 5 seconds, after rainbow screen i have black screen. I tried older version but i have this same problem. How can i fix it?

fydeos-alpha, 2019-10-23:
we can't reproduce the issue you mentioned here. The rainbow screen is normal and expected. Consider changing a new SD card or a new power source see if it helps.

If you can provide more meaningful info or serial logs showing where it failed, it's unlikely that we can be any more helpful.

kondziu95, 2019-11-06:
I had same problem. In my case it occured when i tried to boot system from usb stick. Boot from sd card solved this problem.

closed 2019-12-04 by fydeos-alpha