FydeOS/chromium_os-raspberry_pi #36: widevine DRM

kiai72, 2019-09-06:
does anybody know, how to install libwidevine.so on this system ?

i know about this installscript https://gist.github.com/ruario/19a28d98d29d34ec9b184c42e5f8bf29
This works for chromium-browser on raspbian and archlinux. But not for this FydeOS.

Niclan1, 2019-10-12:
Did not check the existence of the library.

Can however confirm that the browser still sees no drm. In other words, even if it exists it does not function.

kiai72, 2019-10-14:
> Did not check the existence of the library.
> 
> Can however confirm that the browser still sees no drm. In other words, even if it exists it does not function.

any idea, how to fix this ?

yunshansimon, 2019-10-14:
You need to modify the chromium's code at [widevine](https://cs.chromium.org/chromium/src/third_party/widevine/cdm) , then build the chromium.

Niclan1, 2019-10-14:
> > Did not check the existence of the library.
> > Can however confirm that the browser still sees no drm. In other words, even if it exists it does not function.
> 
> any idea, how to fix this ?

Unfortunately not. I was hoping your version of chromium would solve my quest for this library. I will come back here as soon as I have a working libwidevine.so

closed 2019-10-23 by fydeos-alpha

FiodorTretyakov, 2020-08-13:
Hi, any updates on it? Can you include the fix for chromium to your build?