FydeOS/chromium_os-raspberry_pi #41: HDMI audio not working after command

Piflyer, 2019-11-23:
So I followed issue #17, but I still can't force audio via HDMI on my Raspberry Pi 4 on the latest release. Is there a way that I can replace the current file with the default one?

yunshansimon, 2019-12-11:
First, make sure your HDMI audio is working with raspbian.  Then, modify the file `/etc/init/force_audio_output_to_headphones.conf` . If `2` doesn't work, change it to `0`.

sgtram, 2020-03-26:
Hi, whatever I set 2 or 0 the sound always headphones. No HDMI audio. Any idea?

Dittizardichu, 2020-06-05:
I cannot edit the force audio output file from within Chromium OS. It says the file is unwritable, even with root privileges.

closed 2020-12-12 by fydeos-alpha

Fusion3Design, 2020-12-14:
i do not even see this file if I look in /etc/init on the latest build

rimonmath, 2021-10-20:
> i do not even see this file if I look in /etc/init on the latest build

I also don't see this file