FydeOS/chromium_os-raspberry_pi #43: Disable display compositor or switch to xrender

Discombobulated88, 2019-12-07:
Hello ,
I'm wondering if it is possible to disable screen compositing in this Chromium OS or just in chrome os in general.

The gpu compositing that is in place is causing graphical glitches and slow performance on these images on the rpi 4. 

The raspberry pi 4 does not seem to currently work properly with opengl / GLX compositing and it runs wayyy better with it disabled if you can handle a little bit of tearing ( which is happening anyway with the current compositor) 

So is it possible to disable compositing or switch it from OpenGL to xrender instead?

Thank you.

thiago-monteiro, 2019-12-08:
Thank you for mentioning this. I have the same issue. It's bearable but very annoying.

yunshansimon, 2019-12-11:
The recent version of Chromium OS based on "freon" which is a graphic stack combined of  DRI/GBM/OpenGL ES(or VULKAN). 
We use the latest raspberry pi official kernel. If the graphic driver in the kernel doesn't provide stable and efficient performance, we can do nothing about it.

Discombobulated88, 2019-12-11:
Hello , I'm not talking about the graphics driver in general I'm talking about opengl based compositors specifically.   For example the raspberry pi 4 has the same problem in Linux when using the "Compton" opengl compositor or any opengl compositor.   I'm asking you if it is possible to disable the opengl compositor in chrome os.

yunshansimon, 2019-12-11:
@Discombobulated88 Chromium os neet hardware acceleration, there are no other options. I'm sad as you were. When I got the brand new rpi4, I thought it will get the same performance as  rk3288, even better. But at last, It's worst than rpi3. We need more patience to wait for the new driver.

Discombobulated88, 2019-12-11:
It can still have hardware acceleration but the window manager compositor needs to be changed to xrender instead if opengl

fydeos-alpha, 2020-02-14:
Deviation of the roadmap of vanilla Chromium OS is not the intention of this project, especially on the graphics stack which is a huge undertaking. Sorry friend, if you want to re-marry Chromium OS with X you are probably on your own. 

Closing the issue.

closed 2020-02-14 by fydeos-alpha