FydeOS/chromium_os-raspberry_pi #44: Build chromium OS with local browser

killmind, 2019-12-09:
I noticed that in file `overlay-rpi3/make.conf` the way of building browser is set to local.
`CHROME_ORIGIN=LOCAL_SOURCE`. 
Then I cloned a copy of chrome source code and put it in, and sync the code by command 
```
git clone https://chromium.googlesource.com/chromium/src.git
sync_chrome --tag=77.0.3865.126 --reset  $brower_path
```
By restarting build_parckage, cros_sdk found the browser souce code, but reported an error like this:
```
=== Start output for job chromeos-chrome-77.0.3865.126_rc-r1 (0m3.0s) ===
chromeos-chrome-77.0.3865.126_rc-r1: >>> Emerging (1 of 1) chromeos-base/chromeos-chrome-77.0.3865.126_rc-r1::chipset-bcm2837 for /build/rpi3/
chromeos-chrome-77.0.3865.126_rc-r1:  * Running stacked hooks for pre_pkg_setup
chromeos-chrome-77.0.3865.126_rc-r1:  *    sysroot_build_bin_dir ...                                          [ ok ]
chromeos-chrome-77.0.3865.126_rc-r1:  * Running stacked hooks for post_pkg_setup
chromeos-chrome-77.0.3865.126_rc-r1:  *    python_eclass_hack ...                                             [ ok ]
chromeos-chrome-77.0.3865.126_rc-r1:  * Running stacked hooks for pre_src_unpack
chromeos-chrome-77.0.3865.126_rc-r1:  *    python_multilib_setup ...                                          [ ok ]
chromeos-chrome-77.0.3865.126_rc-r1: >>> Unpacking source...
chromeos-chrome-77.0.3865.126_rc-r1: 
chromeos-chrome-77.0.3865.126_rc-r1:  * If you want to develop or hack on the browser itself, you should follow the
chromeos-chrome-77.0.3865.126_rc-r1:  * simple chrome workflow instead of using emerge:
chromeos-chrome-77.0.3865.126_rc-r1:  * https://chromium.googlesource.com/chromiumos/docs/+/master/simple_chrome_workflow.md
chromeos-chrome-77.0.3865.126_rc-r1: 
chromeos-chrome-77.0.3865.126_rc-r1:  * CHROME_ORIGIN VALUE is LOCAL_SOURCE
chromeos-chrome-77.0.3865.126_rc-r1: Building Chromium with additional media codecs and containers.
chromeos-chrome-77.0.3865.126_rc-r1:  * Using GN from /home/hao/work/chrome_root/src/buildtools/linux64/gn
chromeos-chrome-77.0.3865.126_rc-r1: >>> Source unpacked in /build/rpi3/tmp/portage/chromeos-base/chromeos-chrome-77.0.3865.126_rc-r1/work
chromeos-chrome-77.0.3865.126_rc-r1:  * Running stacked hooks for post_src_unpack
chromeos-chrome-77.0.3865.126_rc-r1:  *    asan_init ...                                                      [ ok ]
chromeos-chrome-77.0.3865.126_rc-r1:  * Running stacked hooks for pre_src_prepare
chromeos-chrome-77.0.3865.126_rc-r1:  *    enable_cxx_exceptions ...                                          [ ok ]
chromeos-chrome-77.0.3865.126_rc-r1: >>> Preparing source in /build/rpi3/tmp/portage/chromeos-base/chromeos-chrome-77.0.3865.126_rc-r1/work/chromeos-chrome-77.0.3865.126_rc ...
chromeos-chrome-77.0.3865.126_rc-r1:  * /home/hao/work/chrome_root should be set here properly
chromeos-chrome-77.0.3865.126_rc-r1:  * Applying widevine_gni.patch ...                                       [ ok ]
chromeos-chrome-77.0.3865.126_rc-r1: >>> Source prepared.
chromeos-chrome-77.0.3865.126_rc-r1: >>> Configuring source in /build/rpi3/tmp/portage/chromeos-base/chromeos-chrome-77.0.3865.126_rc-r1/work/chromeos-chrome-77.0.3865.126_rc ...
chromeos-chrome-77.0.3865.126_rc-r1:  * Building with the compiler settings:
chromeos-chrome-77.0.3865.126_rc-r1:  *   CFLAGS = -pipe -fno-split-dwarf-inlining -faddrsig -Wno-unknown-warning-option
chromeos-chrome-77.0.3865.126_rc-r1:  *   CXXFLAGS = -pipe -fno-split-dwarf-inlining -D__google_stl_debug_vector=1 -faddrsig -Wno-unknown-warning-option -stdlib=libc++
chromeos-chrome-77.0.3865.126_rc-r1:  *   CPPFLAGS = 
chromeos-chrome-77.0.3865.126_rc-r1:  *   LDFLAGS = -Wl,-O2 -Wl,--as-needed -stdlib=libc++
chromeos-chrome-77.0.3865.126_rc-r1: /mnt/host/depot_tools/gclient runhooks --force
chromeos-chrome-77.0.3865.126_rc-r1: Error: client not configured; see 'gclient config'
chromeos-chrome-77.0.3865.126_rc-r1:  * ERROR: chromeos-base/chromeos-chrome-77.0.3865.126_rc-r1::chipset-bcm2837 failed (configure phase):
chromeos-chrome-77.0.3865.126_rc-r1:  *   (no error message)
chromeos-chrome-77.0.3865.126_rc-r1:  * 
chromeos-chrome-77.0.3865.126_rc-r1:  * Call stack:
chromeos-chrome-77.0.3865.126_rc-r1:  *     ebuild.sh, line 124:  Called src_configure
chromeos-chrome-77.0.3865.126_rc-r1:  *   environment, line 4503:  Called die
chromeos-chrome-77.0.3865.126_rc-r1:  * The specific snippet of code:
chromeos-chrome-77.0.3865.126_rc-r1:  *           CFLAGS="${CFLAGS} ${EBUILD_CFLAGS[*]}" CXXFLAGS="${CXXFLAGS} ${EBUILD_CXXFLAGS[*]}" LDFLAGS="${LDFLAGS} ${EBUILD_LDFLAGS[*]}" "${cmd[@]}" || die;
chromeos-chrome-77.0.3865.126_rc-r1:  * 
chromeos-chrome-77.0.3865.126_rc-r1:  * If you need support, post the output of `emerge --info '=chromeos-base/chromeos-chrome-77.0.3865.126_rc-r1::chipset-bcm2837'`,
chromeos-chrome-77.0.3865.126_rc-r1:  * the complete build log and the output of `emerge -pqv '=chromeos-base/chromeos-chrome-77.0.3865.126_rc-r1::chipset-bcm2837'`.
chromeos-chrome-77.0.3865.126_rc-r1:  * The complete build log is located at '/build/rpi3/tmp/portage/logs/chromeos-base:chromeos-chrome-77.0.3865.126_rc-r1:20191209-015843.log'.
chromeos-chrome-77.0.3865.126_rc-r1:  * For convenience, a symlink to the build log is located at '/build/rpi3/tmp/portage/chromeos-base/chromeos-chrome-77.0.3865.126_rc-r1/temp/build.log'.
chromeos-chrome-77.0.3865.126_rc-r1:  * The ebuild environment file is located at '/build/rpi3/tmp/portage/chromeos-base/chromeos-chrome-77.0.3865.126_rc-r1/temp/environment'.
chromeos-chrome-77.0.3865.126_rc-r1:  * Working directory: '/build/rpi3/tmp/portage/chromeos-base/chromeos-chrome-77.0.3865.126_rc-r1/work/chromeos-chrome-77.0.3865.126_rc'
chromeos-chrome-77.0.3865.126_rc-r1:  * S: '/build/rpi3/tmp/portage/chromeos-base/chromeos-chrome-77.0.3865.126_rc-r1/work/chromeos-chrome-77.0.3865.126_rc'
chromeos-chrome-77.0.3865.126_rc-r1: >>> Failed to emerge chromeos-base/chromeos-chrome-77.0.3865.126_rc-r1 for /build/rpi3/, Log file:
chromeos-chrome-77.0.3865.126_rc-r1: >>>  '/build/rpi3/tmp/portage/logs/chromeos-base:chromeos-chrome-77.0.3865.126_rc-r1:20191209-015843.log'
```
Do I need to set any other configuration to build the OS together with browser? It's not a problem for me whether CHROME_ORIGIN is LOCAL_SOURCE or from server. I simply want to build a image by myself. Thanks!

killmind, 2019-12-09:
@fydeos-alpha Could you help, sir?

zdm110, 2019-12-10:
I encount the same problem.

yunshansimon, 2019-12-11:
Remove `CHROME_ORIGIN=LOCAL_SOURCE` from `make.conf` will make it easier. We can't build Chrome from the source code at google server for the "GFW" issue.

killmind, 2019-12-11:
Removed `CHROME_ORIGIN=LOCAL_SOURCE`. It started to download source code from google server. But it did start to build without complicated configuation and syncing.

yunshansimon, 2019-12-11:
@killmind what's your meaning of "build without complicated configuration and syncing"?

killmind, 2019-12-11:
When you clone the repository of chrome brower locally, it seems you would need to run `sync_chrome` manually to that local copy. I'm still investigating it.

haiyanghaiyang, 2019-12-25:
I think the guide should be integral with everything inside to avoid confusion for beginners.
The raspberry_pi overy uses local source, which needs download chromium code first, and mount with cros_sdk.

To get chromium code

```
fetch --nohooks chromium
```

If it is broken, run this to continue or sync later.

```
gclient sync
```

There's problem to get WebRTC branch, and you may have to add this into ~/.gitconfig

```
[remote "origin"]
        fetch = +refs/branch-heads/*:refs/branch-heads/*
```

To sync with specific tag to match with raspbarry_pi overlay like r77, you need do this in chromium code, after first code syncing.

```
cd src
git checkout 77.0.3865.126
cd ..
gclient sync
```

After syncing, adding this line to end of .gclient in parent of src directory

```
target_os = ['chromeos']
```

Then you run cros_sdk to mount local chromium source code

```
cros_sdk --chrome-root ${chromium_path}
```

In cros_sdk environment, you could see chromium code mounted at /home/${your_account}/chrome_root directory. Then you can run build_packages or use "emerge-${BOARD} chromeos-base/chromeos-chrome" to build chromium.

fydeos-alpha, 2020-01-16:
Instructions added to README. Thank you @haiyanghaiyang