FydeOS/chromium_os-raspberry_pi #46: No HDMI Out on Raspberry Pi 4

InteractiveNinja, 2019-12-18:
Hi there, 

I tried to install the latest Build for the PI 4 but I dont get any Signal from the HDMI.
The Screen wakes up on Boot but it changes to no Signal.

I think I had it running for a Short time on my First try but I dont got it there anymore.

yunshansimon, 2019-12-20:
How about booting with Rasbian?

InteractiveNinja, 2019-12-20:
Right now I am only capable of Booting Rasbian when I put hdmi_safe=1 in the config.txt.

I think my Monitor could be the Issue.  I will try to switch to a other one.

I tried to change the config.txt from Chromium OS but it is write Protected or something...
Got any Idea how change config.txt the values from the Bootable SD Card?

yunshansimon, 2019-12-20:
```
mkdir /tmp/boot
sudo mount /dev/mmcblk0p12 /tmp/boot
sudo vi /tmp/boot/config.txt
```

InteractiveNinja, 2019-12-20:
Are this Commands ment to be used in raspbian or how can I use them?

yunshansimon, 2019-12-20:
Sorry about the script. It runs under Chromium OS actually. You can mount the 12th partition on the SD card under any Linux distro. the `config.txt` on the partition is as same as raspbian. You can edit with `root` privilege according to the official document for raspbian.

InteractiveNinja, 2019-12-20:
I see, now I am asking is there any way to do it in Windows 10? 

Its not a problem for me to install a VM for that but just want to ask for the future, is there a other way to do it?

InteractiveNinja, 2019-12-21:
Well I got it running. My Monitor was the Problem. Thanks @yunshansimon for the Info how to edit the config.txt.

I added  **hdmi_ignore_edid=0xa5000080** to the Config for "bypassing" the Problem.

Thanks for the fast replays.

W9JYD, 2020-01-06:
Nope. Not working for me. I have tried **everything**. Including the `hdmi_ignore_edid=0xa5000080` line. I have my own `config.txt` I use for *every* RasPi SD Card. Works in Raspbian. Why not this?

My config.txt file:
```
# Customized 2019-12-10 W9JYD (https://www.qrz.com/db/W9JYD)
# 
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
overscan_left=-12
overscan_right=-12
overscan_top=-20
overscan_bottom=-20

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=1
hdmi_mode=16

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
hdmi_drive=2
#hope this shit works...
hdmi_ignore_edid=0xa5000080

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
config_hdmi_boost=11

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
#max_framebuffers=2
#FUCK vc4-fkms-3d!!!!

[all]
#dtoverlay=vc4-fkms-v3d
#I fucking said fuck "vc4-fkms-3d"! What part of # you don't understand??

gpu_mem=160
core_freq_min=500
```
What is wrong with this? It boots Raspbian Buster fine!

InteractiveNinja, 2020-01-28:
Well my problem got fixed, wish you luck @W9JYD

closed 2020-01-28 by InteractiveNinja