FydeOS/chromium_os-raspberry_pi #47: Error when I tried to install chromium with a family link account

Purpleman666, 2019-12-22:
Hello guys, 

Thank you for your awsome work.
I've tried to install chromium r77r2 on a raspberry 4. I's perfect with a normal account, but it don't work if you tried with a google family link account. It begin well, but at the end it encounted an error, and it says that i have to do it another time. I tried with another family link account it's the same thing.

faizhasim, 2020-03-29:
I encounter same issue. I am suspecting family link won't be supported. 

Regardless, work around is, you can use guess account - although that's definitely lack personalisation.

dawithers, 2020-04-24:
Cloudready has the same issue and they say it's because it requires a TPM chip to be available in the device, which obviously this doesn't have. So it may not even be possible.

https://neverware.zendesk.com/hc/en-us/community/posts/206761367-Unable-to-create-Supervised-User

SidShetye, 2020-04-24:
If that’s confirmed to be the case, then the TPM’s just a device. Can be emulated to workaround this issue. In fact, there are a couple of projects. First google hit https://github.com/PeterHuewe/tpm-emulator ... I wish I had the time to investigate myself but seems like a fun project.

closed 2020-12-12 by fydeos-alpha