FydeOS/chromium_os-raspberry_pi #49: Touchscreen Inverse X

pt4si3k83, 2020-01-11:
Hi,
I successfully installed the latest build on Rpi4 and I'm using 22inches LCD monitor with USB touchscreen. Touchscreen is being detected by system as 'XDTouch IrTouchScreen' . Everything works fine BUT its X axis is inverted. I spent alot of time trying to do so, but I don't know "freon" at all. 

If there was an XOrg then adding the following part to 'xorg.conf' would do the trick:
`Section "InputDevice"
	 Identifier "touchscreen"
	 Driver "evdev"
	(...)
	 Option "InvertX" "True"
EndSection`

 Is there any chance to configure it?

yunshansimon, 2020-01-13:
The input device configuration files are at `/etc/gesture`.

pt4si3k83, 2020-01-13:
Than you @yunshansimon . I'm trying to configure it in the same way (or similar) I did on Raspbian, but it seems to not working at all. Can you tell me where can I find UI logs so that I could see whether my configuration is being read at all? Is there any log file - like 'xorg.log' ?

yunshansimon, 2020-01-14:
`/var/log/ui/ui.*` :  graphic related logs. 
`/var/log/chrome/chrome` : chrome main logs.
`evtest` : the command for testing input device.

pt4si3k83, 2020-01-19:
Hey, I tried to modify one of the file which is already in `/etc/gestures`. I tried to create a new one there with my configuration. Unfortunatelly there were no change in the system nor any information in any log file. 

Are the `/etc/gestures/*.conf` files being used at all?

pt4si3k83, 2020-01-22:
Is there any way to recompile the kernel module (usbtouchscreen.c) on my RPi4 to add a support for "InvertX" parameter? I was searching for this over the internet but I didn't find any useful information (instructions).

pt4si3k83, 2020-01-27:
I managed to rebuild `usbtouchscreen` kernel module to support `invert_x` parameter and now it works fine. My issue with X axis seems to be solved.

However I think there still might be something wrong with `/etc/gestures/*.conf` files support.

closed 2020-01-27 by pt4si3k83