FydeOS/chromium_os-raspberry_pi #51: [Issue report]

IanKnighton, 2020-01-17:
**Describe the bug or issue**
It's entirely possible that I'm missing something, but when I use the latest release it just boots into a kiosk mode. I couldn't find a way out of it in order to do an enterprise enrollment. 

**Describe your setup**
- Raspberry Pi 3b
- [Latest Release](https://github.com/FydeOS/chromium_os-raspberry_pi/releases/tag/r77-kiosk)

**To Reproduce**
Fresh install using Etcher. Boot screen looks similar to older images, but post boot is just a periodic table web page that I can't get past. 

**Expected behaviour**
I would expect the bootup to behave like the previous releases and allow me to login or attempt an enterprise enrollment.

yunshansimon, 2020-01-18:
You can disable this feature by removing the flag `--force-kiosk-mode` in `/etc/chrome_dev.conf` at ROOT-A partition.

fydeos-alpha, 2020-01-20:
Release note updated on this. It's a feature rather than a bug, issue closed.

closed 2020-01-20 by fydeos-alpha