FydeOS/chromium_os-raspberry_pi #52: Timezone Resets on Reboot

IanKnighton, 2020-01-22:
**Describe the bug or issue**
I've set the Raspberry Pi up as a Kiosk to display our dashboard from Uptime Robot. On launch, I see a screen that says "Clock is behind" and no pages with SSL will load. 

I've found that I can go into the command line and run `sudo date -s "Wed Jan 22 20:59:00 UTC 2020"` and then refresh the page, but that isn't really tenable for a standalone kiosk. 

I haven't found a way to set this and have it persist after reboot. 

I've also found that, even as root, I am unable to edit the `/etc/chrome_dev.conf` file to turn off kiosk mode and try to set this through the Chrome OS. 

**Describe your setup**
- Raspberry Pi 3b
- Connected via HDMI to a TV

**To Reproduce**
Launch into Kiosk Mode with a page using SSL

**Expected behaviour**
The timezone should be able to be set and persisted across reboots.

IanKnighton, 2020-01-22:
So I've tried to change the `/etc/localtime' link to make sure it's in my timezone. I'm running into another error that says it's a readonly filesystem. 

```
chronos@localhost ~ $ sudo ln -sf /usr/share/zoneinfo/America/Boise /etc/localtime 
ln: cannot remove '/etc/localtime': Read-only file system
```

Did I do something wrong when I flashed the SD card?

fydeos-alpha, 2020-01-22:
To modify rootfs you will need to issue this command first:
```
sudo mount -o remount rw /
```

We are still investigating the time issue.

josedelinux, 2020-02-02:
[image]
same issues .
I also noticed that it tried to access a website( periodex.co ) about periodic table of elements,is it a test of connection? 
by the way ,how to open terminal? , I tried both `ctrl`+`alt`+`T` and `ctrl`+`alt`+`f2`, not working

yunshansimon, 2020-02-24:
try:

```bash
$ sudo mount -o remount rw /
$ sudo sh -c 'echo start on started boot-services > /etc/init/system-services.override'
$ sudo reboot
```

fydeos-alpha, 2020-05-07:
resolved in r80 release, closing.

closed 2020-05-07 by fydeos-alpha