FydeOS/chromium_os-raspberry_pi #53: [Issue report]

Meowcat285, 2020-01-27:
**Describe the bug or issue**
A clear and concise description of what the bug is.

**Describe your setup**
 - Your Raspberry Pi model and spec variant, e.g. 4B with 2GB RAM
 - Your Chromium OS for Raspberry Pi release version, e.g. Chromium OS r77 r3
 - Your hardware configuration and setup, e.g. how you have wired the Pi board with any other peripheral connected to it

Raspberry Pi 4B with 4 GB of RAM
Chromium os r77 R2

**To Reproduce**
Steps to reproduce the behavior:
1. Start system
2. Animations flicker

**Expected behaviour**
Animations are smooth

**Screenshots and/or hardware setup photos**
If applicable, add screenshots to help explain your problem. If your problem is hardware related, it's much easier to include a photo of how your setup looks like.

N/A
**Additional context**
Add any other context about the problem here.

N/A

fydeos-alpha, 2020-01-31:
We are going to need a bit more information regarding your physical setup, did you use any HDMI converter? What kind of screen? A picture/video demonstrating the flicker would be ideal.

Meowcat285, 2020-02-01:
I am not using a converted, it is a LG 1080p display. I saw another issue regarding the same issue from a month or so ago

LucasFidelis, 2020-02-14:
I have the same problem with a screen E230t HP without HDMI converter.

jwnola, 2020-08-20:
I'm using the latest prebuilt image, and I'm having the same issues where the screen goes black randomly. I'm using a Raspberry Pi 4 and a Dell 27 running at 1920x1080 directly connected with a micro HDMI to HDMI cable. No converter.

Sewdn, 2020-11-02:
I'm using the latest prebuilt image as well. Extreme flickering and noisy display on 4K resolution (using a Philips Brilliance 4K screen), without any adapter. Using a micro HDMI to HDMI cable ().
Less flickering, but very frequent blackouts on a (old) LG FullHD screen 1920x1080.

Related to https://github.com/FydeOS/chromium_os-raspberry_pi/issues/84

th0mpy, 2020-11-06:
Same here, using a HDMI cable directly to the RPi without an adapter. I get a lot of black/blank screens. The cursor is "corrupted" as well. RPi 4 8g model.

fydeos-alpha, 2020-12-12:
flickering is now much improved in the r86r1 build, give it a try

closed 2020-12-12 by fydeos-alpha

a9udn9u, 2021-02-24:
For people having the same problem, increase gpu mem solved the issue for me.

Follow [this post](https://github.com/FydeOS/chromium_os-raspberry_pi/issues/109#issuecomment-739442120) to mount the boot partition, update config.txt, add `gpu_mem=256`, reboot.

128MB may also work but I didn't bother trying.

Bituvo, 2021-12-11:
I'm having this issue with my Github Pages website, [here](https://bituvo.github.io/)
Animations flicker and turn black.