FydeOS/chromium_os-raspberry_pi #57: Filesystem /etc/ write-protected

tomrow, 2020-02-11:
I can't disable kiosk mode due to a write-protected partition, how can I enable writing to chrome_dev.conf?

fydeos-alpha, 2020-02-14:
do this:
```
sudo mount -o remount rw /
```

Then you will have read-write access of `/`.

ifzl, 2021-04-12:
I have managed to do the following steps but still something is wrong or missing;
Step 1 =  CRTL+ALT+T  - this will open the Crosh window
Step 2 =  use the command:   shell     ( this will open the shell ) 
Step 3 =  use the command:   sudo mount -o remount rw /    ( this will give you the writing rights ) 
Step 4 =  use the command:   sudo vi /etc/chrome_dev.conf    ( this will bring you to the file you want to edit ) 
Step 5 =  add the line:    --force-kiosk-mode
Step 6 =  press Esc
Step 7 =  type:       :wq!    ( and then press Enter)
Step 8 =  sudo reboot 

So far this has not given me the result I am looking for.  Can someone indicate what I should change ?