FydeOS/chromium_os-raspberry_pi #59: Solved - No video output with Elecrow RC070 7 inch 1024*600 HDMI LCD Display with Touch Screen

darko-mijic, 2020-02-25:
**Describe the bug or issue**
There is no video output with this display connected to the HDMI port.

[image]

**Describe your setup**
 - Raspberry Pi 4B with 4GB RAM
 - Chromium OS for Raspberry Pi release version, e.g. Chromium OS r77 r2
 - Elecrow RC070 7 inch 1024*600 HDMI LCD Display with Touch Screen
https://www.elecrow.com/7-inch-1024-600-hdmi-lcd-display-with-touch-screen.html

**To Reproduce**
1. Flash image on SD card
2. Boot Chromium OS

**Solution**

1. SSH from another computer
`ssh chronos@192.168.X.X``
2. Enter password: `test0000`
3. Edit configuration file: 
```
sudo su
mount -o remount rw /
mkdir /tmp/p12
mount /dev/mmcblk0p12 /tmp/p12

vi /tmp/p12/config.txt    
```
4. Add settings recommended by the manufacturer to `config.txt`:
```hdmi_force_hotplug=1
max_usb_current=1
hdmi_group=2
hdmi_mode=1
hdmi_mode=87
hdmi_cvt=1024 600 60 3 0 0 0
hdmi_drive=1
```

5. Unmount and reboot
```
umount /tmp/p12
reboot
```

darko-mijic, 2020-02-25:
I would like to use this display in portrait mode. I have tried with `display_rotate=1` but I get a flickering image.

darko-mijic, 2020-02-25:
Also, the display is presented as **Unknown Display** in Settings and there are no controls.

fydeos-alpha, 2020-12-12:
dup #109

closed 2020-12-12 by fydeos-alpha