FydeOS/chromium_os-raspberry_pi #60: Booting from USB card /dev/sda possible?

vpelss, 2020-02-26:
Is booting from USB card /dev/sda , and not an SD card, possible on the PI version of ChromeOS? 

With raspbian you just modify the config.txt in the boot partition. But with the ChromeOS more complex partition layout it looks like the PI boot files and settings are locked away in ROOT-A/boot/zImage-4.19.83-v7-g3c235dcfe80a-dirty ? 

Can it be done if I attempt to build my own ChromeOS from this source? (I am attempting it, but it is quite a steep learning curve)

vpelss, 2020-02-26:
This can be closed. Linux OS was not picking up the Fat partition by defaukt. When the usb was inserted in a Windows Os, the fat partition showed and I could edit the cmdline. txt to change root=/dev/sda3

closed 2020-02-26 by vpelss

mg1234567, 2020-02-27:
> This can be closed. Linux OS was not picking up the Fat partition by defaukt. When the usb was inserted in a Windows Os, the fat partition showed and I could edit the cmdline. txt to change root=/dev/sda3

Can you specify what you did, I'm trying to boot Chromium from the USB 3 port of the Pi4b.  
Following this vid @8:50
https://www.youtube.com/watch?v=FM9wuFLufyA

I edited cmdline.txt 

changed root=/dev/sda3

But, when I go to boot I get a an error
"end kernel panic - not syncing: VPS: Unable to mount root fs on unknown-block(0,0)"

I've tried sda1 - sda7

Any help would be greatly appreciated

vpelss, 2020-02-27:
Is it possible on a PI 4?
See: 
https://jamesachambers.com/raspberry-pi-4-usb-boot-config-guide-for-ssd-flash-drives/

This was required on the PI 3, but it might cause damage on a Pi4
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md