FydeOS/chromium_os-raspberry_pi #61: Pi4b FydeOS via USB - is it possible.  Any advice / help?

mg1234567, 2020-03-03:
Raspberry Pi 4b 4gb 
Sandisk USB 3.1 thumb drive. 

I've been messing around with this for a while but haven't been able to get it to work.  

Yes, I can boot Rasbian from a USB by messing with the SD card. 
See:
https://jamesachambers.com/raspberry-pi-4-usb-boot-config-guide-for-ssd-flash-drives/

But, I can't seem to get FlydeOS to boot. 
If anyone has any expertise and wants to share a step by step walk-through, I'd greatly appreciate it. 

Thanks!

rhildred, 2020-06-20:
I would also like to do something like this. Ideallly use the SSD like a "live" and give the option to install on to a usb device.

fydeos-alpha, 2020-12-12:
It's out of scope for the purpose of this project, sorry mate.

closed 2020-12-12 by fydeos-alpha