FydeOS/chromium_os-raspberry_pi #63: Use all SD memory

bronco401, 2020-03-23:
Sorry, this is probably a dumb question, but when I get everything setup on my SD card and run it on my Pi, is states there are only a few MB free.  How to I reallocate the unused space on the SD card so FydeOS can use it?

The OS launches fine, but I can't get my profile to download as it runs out of memory.

3B with 32gb SD - I've very new to the Pi and I'm trying to setup a chrome desktop for my kids for school now that their home.

markwhi, 2020-03-27:
Take a look at Issue #10 for the fix.

bronco401, 2020-04-04:
I still don't get how to access the terminal?  Sorry I'm a newbie.

fydeos-alpha, 2020-05-07:
Enter tty shell by hitting ctrl + alt + F2, log in with username `chronos`, password `test0000`

yetipi3, 2020-07-23:
ill be honest i don't understand very much either, any help would be great

Jerry8538, 2020-09-28:
@fydeos-alpha @bronco401 ctrl + alt + t works too. If your problem is solved, you can close this issue.

closed 2020-12-12 by fydeos-alpha