FydeOS/chromium_os-raspberry_pi #65: How to disable kiosk mode

nutellax, 2020-04-01:
I don't know how access /etc/chrome_dev.conf to disable kiosk mode. Sorry if these seems obvious.

kjslack, 2020-04-07:
I am having the same difficulty. How do I get access to a text editor on ChromeOS in Kiosk mode so I can edit the specified file and remove that flag?

davetustin, 2020-04-11:
I would also like to know the answer to this also. How do we go about editing the conf file to disable kiosk mode?

salehram, 2020-04-14:
Hello all,
So this is how I did work it out. Not sure if this is one of the proper ways to do this, but the system did boot into the desktop environment after that.

I had to use a Linux (any Linux will work). I mounted the SD card on that and opened the Root-A partition and just used a normal text editor (kwrite, gedit... whatever) to modify the file and comment the flag they mentioned. I made sure the file is saved, and just to be sure I removed the SD card and inserted it again and double checked the contents of the chrome_dev.conf file to confirm the changes are saved.

Then I just put it back into the Raspberry Pi and booted manually.

The filesystem on the SD card seems to be ext2 and I have no idea how to access that from Windows.

fydeos-alpha, 2020-05-07:
Kiosk mode has been disabled by default in the r80 release and onwards.

closed 2020-05-07 by fydeos-alpha