FydeOS/chromium_os-raspberry_pi #69: r80 on rpi 4B graphical glitches even after login

dawithers, 2020-04-24:
There are lots of graphical glitches with this latest load even after logging in. The release notes sort of imply that the glitches only happen before login, but that doesn't seem to be the case. When typing in text fields, the cursor frequently glitches back and forth and text appears and disappears as you type. When switching tabs in chrome the screen flickers frequently. Video seems to play without issue and the screen appears fine at rest.

Setup
 - 4B with 4gb ram
 - r80 - 6285f66 from releases page - chromiumos_test_image_r80r1-rpi4b.img.xz
 - Connected to Asus monitor via HDMI

chathil, 2020-05-13:
i'm having the same issue.

ololmarc, 2020-05-28:
Having same issue. Updated to 5.4 kernel as well. Perhaps it is the HDMI?

SrIzan10, 2020-06-17:
I have the same issue but in the 3B+ and 768p screen

fydeos-alpha, 2020-12-12:
please try r86r1 build instead.

closed 2020-12-12 by fydeos-alpha