FydeOS/chromium_os-raspberry_pi #72: NaCl support?

Botspot, 2020-05-30:
Many Chrome Apps don't work in this FydeOS. Specifically Zoom and VNC.
Reason: NaCl.
Many RPi users want some a Zoom client, and the most viable option they have is the Chrome App.
Too bad it doesn't work, just because NaCl is not implemented!

Countless users would thank you if you released a version that had NaCl.
Ref: https://www.raspberrypi.org/forums/viewtopic.php?f=131&t=269833&p=1643325#p1643325

fydeos-alpha, 2020-08-02:
this is now solved in the latest r83 release.

closed 2020-08-02 by fydeos-alpha

Botspot, 2020-08-05:
Great to hear!
Some people will prefer to use Raspbian's Chromium to do Zoom calls. **How difficult would it be to port your Chromium executable to Raspbian?**

rahulraj80, 2020-10-30:
> Great to hear!
> Some people will prefer to use Raspbian's Chromium to do Zoom calls. **How difficult would it be to port your Chromium executable to Raspbian?**

That would be a god send!

rahulraj80, 2020-11-02:
That was a very awesome thing @Botspot : We sorely needed it. I had tried it based on your comments at https://devforum.zoom.us/t/have-you-considered-building-zoom-for-the-raspberry-pi-4-linux-system/14257/11 , however, it is still very choppy and unusable on Pi3 with PiCamera and bluetooth headset. 

Am I doing something wrong, or is there some optimization I can do, specifically for Pi3? (By my estimates, there should be around 15 million Pi 3's out of the over 30 million Raspberry Pi's in the world, so Pi 3 should be a big target segment)

Botspot, 2020-11-02:
I am a bit confused, @rahulraj80 .You said you "tried it" based on my devforum zoom comment link. But the link you sent does not provide any Zoom solution. Instead, it asked the Zoom devs to make a RPi Zoom solution. So what did you try?

As far as the Pi3 goes, there's not much I can do. I have many Pi3s lying around unused, and it makes me feel bad too. But the Pi4 is around 6 times faster for graphics-related stuff, so please consider upgrading for smooth video playback.

If you still want to use your Pi3, there's a few things you can do: overclock, enable ZRAM & swap to double your RAM, stick with a high-end sd card - **don't** use a usb ssd, and most importantly, enable Fake KMS OpenGL. After those 4 tweaks, my Pi3 could handle _Facebook_ fairly smoothly.

rahulraj80, 2020-11-03:
Hi @Botspot :  

With respect to the link, I meant Chills340 and your own hints regarding the Box86 approach. I followed the Novaspirit way first, but got choppy results. I thought I might have missed something like the pulseaudio/alsa thing, so re-flashed, and followed the Pi-Apps process (Should have done this in the very first go, it had no heartburn, unlike the long form method), but got same results.

Will try those 4 steps you shared (ZRAM is already live) and revert if I get stuck.