FydeOS/chromium_os-raspberry_pi #74: Chromium Outdated

SrIzan10, 2020-06-15:
I have a problem with Chromium. It's outdated. I can't play Spotify music because of the outdated Chromium.

 - Raspberry Pi 3B+ with 1GB RAM
 - Release version: Chromium OS 80.0.3987.158 (I am new in the system, so I don't know where search it)
 - I have the Pi board connected to my Router, and the Spanish (Spain) version of Chromium OS becouse I am spanish

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the Spotify App
2. Put some music
[image]

4. See error: 
 [image]
                               **An error has occurred**
Spotify is no longer compatible with this version of Chrome. Update the browser to listen without interruption.

**Expected behaviour**
Put the music normal

fydeos-alpha, 2020-08-02:
could you please retry this on the r83 release, we have tested spotify and it worked fine

FiodorTretyakov, 2020-08-13:
Does it mean the widevine DRM work on r83?

SrIzan10, 2020-08-25:
Oh, sorry for no responding... I am going to try it right now.

closed 2020-12-12 by fydeos-alpha

YuriFontella, 2021-02-19:
I have a similar problem in the r86r2-rpi4b version. I don't know if it can be related to widevine. Can this ever work?

[image]

_"It looks like your browser is no longer compatible with netflix. Install the latest version or try a different browser."_

SrIzan10, 2021-04-04:
Sorry for the no response, now it works.

SrIzan10, 2021-04-04:
> I have a similar problem in the r86r2-rpi4b version. I don't know if it can be related to widevine. Can this ever work?
> 
> [image]
> 
> _"It looks like your browser is no longer compatible with Netflix. Install the latest version or try a different browser."_

This issue had been closed @YuriFontella . Try making another issue.