FydeOS/chromium_os-raspberry_pi #83: The screen is always is distorted on the raspberry pi 3B+ on build r83r1 rpi-3b

LCCR193, 2020-08-08:
My pi 3B+ has display working till the chromium logo/boot screen part, it is fine,but at the setup the screen is a total mess, distorted more than anything, I can click things, and stuff changes, but no way of knowing what is on the screen, I compared my config.txt file from the build, everything is same but, the display is distorted.....
Is there any way to fix this issue?

LCCR193, 2020-08-14:
> My pi 3B+ has display working till the chromium logo/boot screen part, it is fine,but at the setup the screen is a total mess, distorted more than anything, I can click things, and stuff changes, [...]

fydeos-alpha, 2020-12-12:
please try r86r1 instead.

closed 2020-12-12 by fydeos-alpha