FydeOS/chromium_os-raspberry_pi #85: Issue with building

WindowsXP95, 2020-08-09:
I followed the instructions when building and got this

!!! CONFIG_PROTECT is empty for '/build/rpi3/'

These are the packages that would be merged, in order:

Calculating dependencies  ... .... done!

!!! The ebuild selected to satisfy "sys-kernel/upstream-kernel-mainline" for /build/rpi3/ has unmet requirements.
- sys-kernel/upstream-kernel-mainline-5.4.21-r2::chipset-bcm2837 USE="boot_dts_device_tree dt_compression fit_compression_kernel_lz4 lld mbim tpm unibuild vdso32 vlan -acpi_ac -allocator_slab -apex -apply_patches -asan -binder -blkdevram -buildtest -builtin_driver_amdgpu -builtin_fw_amdgpu -builtin_fw_guc_g9 -builtin_fw_guc_jsl -builtin_fw_guc_tgl -builtin_fw_huc_g9 -builtin_fw_huc_jsl -builtin_fw_huc_tgl -builtin_fw_t124_xusb -builtin_fw_t210_bpmp -builtin_fw_t210_nouveau -builtin_fw_t210_xusb -builtin_fw_vega12 -builtin_fw_x86_aml_ucode -builtin_fw_x86_apl_ucode -builtin_fw_x86_bdw_ucode -builtin_fw_x86_bsw_ucode -builtin_fw_x86_byt_ucode -builtin_fw_x86_cml_ucode -builtin_fw_x86_glk_ucode -builtin_fw_x86_jsl_ucode -builtin_fw_x86_kbl_ucode -builtin_fw_x86_skl_ucode -builtin_fw_x86_whl_ucode -ca0132 -cec -clang -compilation_database -criu -cros_ec_mec -cros_host -debug -debugobjects -devdebug -device_tree -diskswap -dm_snapshot -dmadebug -docker -dp_cec -drm_dp_aux_chardev -dwc2_dual_role -dyndebug -ec2_guest_net -eve_bt_hacks -eve_wifi_etsi -factory_netboot_ramfs -factory_shim_ramfs -failslab -fbconsole -firmware_install -fit_compression_kernel_lzma -goldfish -highmem -hypervisor_guest -i2cdev -iscsi -kasan -kcov -kexec_file -kgdb -kmemleak -kvm -kvm_host -kvm_nested -lockdebug -lockdown -memory_debug -module_sign -nested_l3_kvm -nfc -nfs -nowerror -pca954x -pcserial -plan9 -ppp -pvrdebug -qmi -realtekpstor -recovery_ramfs -samsung_serial -selinux_develop -socketmon -systemtap -test -tpm2 -transparent_hugepage -ubsan -usb_gadget -usb_gadget_acm -usb_gadget_audio -usb_gadget_ncm -usbip -vfat -virtio_balloon -vivid -vmware_guest -vtconsole -wifi_diag -wifi_testbed_ap -wireless318 -wireless34 -wireless38 -wireless42 -x32 -xen_guest" BOARD_USE="-acorn -amd64-corei7 -amd64-generic -amd64-generic-cheets -amd64-generic-goofy -amd64-generic_embedded -amd64-host -aplrvp -aries -arkham -arm-generic -arm64-generic -arm64-llvmpipe -asuka -asurada -atlas -atlas-arm64 -atlas-blueznext -atlas-kvm -auron -auron_paine -auron_pearlvalley -auron_yuna -banjo -banon -bayleybay -beaglebone -beaglebone_servo -beaglebone_vv1 -beltino -betty -betty-arc64 -betty-arcmaster -betty-arcnext -betty-arcvm-master -betty-arcvm-pi -betty-pi-arc -blackwall -bob -bobcat -bolt -borealis -bruteus -bubs -buddy -buddy-cfm -butterfly -bwtm2 -candy -capri -capri-zfpga -cardhu -caroline -caroline-arc64 -caroline-kernelnext -caroline-ndktranslation -caroline-userdebug -cave -celes -celes-cheets -chell -chell-cheets -cheza -cheza-freedreno -cheza64 -cheza64-freedreno -cid -clapper -cmlrvp -cobblepot -coral -cosmos -cranky -cyan -cyan-cheets -cyan-kernelnext -cyclone -daisy -daisy_embedded -daisy_skate -daisy_snow -daisy_spring -daisy_winter -dalmore -danger -danger_embedded -dedede -deltaur -drallion -duck -edgar -elm -elm-cheets -elm-kernelnext -endeavour -enguarde -eve -eve-arc-r -eve-arcnext -eve-arcvm -eve-arcvm-mesa-virgl-next -eve-arm64 -eve-campfire -eve-kernelnext -eve-kvm -eve-lacros -eve-manatee -eve-swap -eve-userdebug -excelsior -expresso -falco -falco_gles -falco_li -fb1 -fizz -fizz-accelerator -fizz-cfm -fizz-labstation -fizz-moblab -flapjack -foster -gale -gandof -glados -glados-cheets -glimmer -glimmer-cheets -glkrvp -gnawty -gonzo -gru -grunt -grunt-kernelnext -guado -guado-accelerator -guado-cfm -guado-macrophage -guado_labstation -guado_moblab -hana -hana-kernelnext -hatch -hatch-arc-r -hatch-arc-r-userdebug -hatch-blueznext -hatch-borealis -hatch-diskswap -hatch-kernelnext -heli -hsb -ironhide -jacuzzi -jecht -kalista -kalista-cfm -kayle -kblrvp -kefka -kefka-kernelnext -kevin -kevin-arc64 -kevin-tpm2 -kevin64 -kidd -kip -klang -kukui -kukui-arc-r -kukui-arc64 -kumo -kunimitsu -lakitu -lakitu-gpu -lakitu-nc -lakitu-slice -lakitu-st -lakitu_next -lars -laser -lasilla-ground -lassen -leon -link -littlejoe -loonix -lulu -lulu-cheets -lumpy -mappy -mappy_flashstation -marble -mccloud -metis -minnowboard -mipseb-n32-generic -mipseb-n64-generic -mipseb-o32-generic -mipsel-n32-generic -mipsel-n64-generic -mipsel-o32-generic -mistral -moblab-generic-vm -monroe -moose -mushu -nami -nami-kvm -nautilus -ninja -nocturne -nocturne-arm64 -novato -novato-arc64 -novato-arcnext -nyan -nyan_big -nyan_blaze -nyan_kitty -oak -oak-cheets -octavius -octopus -orchestra -orco -palkia -panda -panther -panther_embedded -panther_goofy -panther_moblab -parrot -parrot32 -parrot64 -parrot_ivb -peach -peach_kirby -peach_pi -peach_pit -peppy -plaso -poppy -ppcbe-32-generic -ppcbe-64-generic -ppcle-32-generic -ppcle-64-generic -puff -puppy -purin -pyro -quawks -rainier -rambi -rammus -rammus-arc-r -raspberrypi -reef -reks -relm -reven -rikku -rikku-cfm -rizer -romer -rotor -rowan -rush -rush_ryu -sama5d3 -samus -samus-cheets -sand -sarien -sarien-kvm -scarlet -sentry -setzer -shadowkeep -shogun -sklrvp -smaug -smaug-cheets -smaug-kasan -snappy -sonic -soraka -squawks -stelvio -storm -storm_nand -stout -strago -stumpy -stumpy_moblab -stumpy_pico -sumo -swanky -tael -tails -tatl -tegra3-generic -terra -tglrvp -tidus -tricky -trogdor -trogdor-kernelnext -ultima -umaro -veyron -veyron_fievel -veyron_gus -veyron_jaq -veyron_jerry -veyron_mickey -veyron_mighty -veyron_minnie -veyron_minnie-cheets -veyron_nicky -veyron_pinky -veyron_remy -veyron_rialto -veyron_shark -veyron_speedy -veyron_speedy-cheets -veyron_thea -veyron_tiger -viking -viking-poc2 -volteer -volteer-kernelnext -whirlwind -whlrvp -winky -wizpig -wolf -wooten -wristpin -wsb -x30evb -x32-generic -x86-agz -x86-alex -x86-alex32 -x86-alex32_he -x86-alex_he -x86-alex_hubble -x86-dogfood -x86-generic -x86-generic_embedded -x86-mario -x86-mario64 -x86-zgb -x86-zgb32 -x86-zgb32_he -x86-zgb_he -zako -zoombini -zork -zork-kernelnext"

  The following REQUIRED_USE flag constraints are unsatisfied:
    lld? ( clang )

  The above constraints are a subset of the following complete expression:
    compilation_database? ( clang ) fit_compression_kernel_lz4? ( !fit_compression_kernel_lzma ) fit_compression_kernel_lzma? ( !fit_compression_kernel_lz4 ) lld? ( clang ) kgdb? ( vtconsole ) factory_netboot_ramfs? ( !recovery_ramfs !factory_shim_ramfs ) factory_shim_ramfs? ( !recovery_ramfs !factory_netboot_ramfs ) recovery_ramfs? ( !factory_netboot_ramfs !factory_shim_ramfs ) factory_netboot_ramfs? ( i2cdev ) factory_shim_ramfs? ( i2cdev ) recovery_ramfs? ( i2cdev ) factory_netboot_ramfs? ( any-of ( tpm tpm2 ) ) factory_shim_ramfs? ( any-of ( tpm tpm2 ) ) recovery_ramfs? ( any-of ( tpm tpm2 ) )

(dependency required by "virtual/linux-sources-1-r21::chromiumos[kernel-upstream-mainline]" [ebuild])
(dependency required by "net-vpn/strongswan-5.7.2-r2::chromiumos" [ebuild])
(dependency required by "chromeos-base/vpn-manager-0.0.1-r2730::chromiumos" [ebuild])
(dependency required by "chromeos-base/shill-0.0.4-r2170::chromiumos[vpn]" [ebuild])
(dependency required by "chromeos-base/patchpanel-0.0.1-r74::chromiumos" [ebuild])
(dependency required by "chromeos-base/system-proxy-0.0.1-r115::chromiumos" [ebuild])
(dependency required by "virtual/target-chromium-os-1-r141::chromiumos[system_proxy]" [ebuild])
(dependency required by "virtual/target-os-1-r3::chromiumos" [ebuild])
(dependency required by "virtual/target-os" [argument])
01:07:24 ERROR   : emerge detected broken ebuilds. See error message above.

Any help

yunshansimon, 2020-08-09:
Pls, fetch all tags of the repo, and checkout the tag "r83".

WindowsXP95, 2020-08-09:
ok thanks

closed 2020-12-13 by fydeos-alpha