FydeOS/chromium_os-raspberry_pi #87: [Issue report] Cant connect Raspberry Camera Pi

rizdam, 2020-08-14:
**Describe the bug or issue**
Cant connect module raspberry camera, when i open camera said "You camera is currently unavaliable"

**Describe your setup**
 - Raspberry Pi 3 Model
 - Chromium OS r83
 - Original Camera Raspberry

**To Reproduce**
Steps to reproduce the behavior:
1. Open Camera

**Expected behaviour**
when i open camera said "You camera is currently unavaliable", i already put my camera

MadTooler, 2020-09-28:
Is there a way to use a pi camera module with Chromium OS?

If not, are there any near future plans to add it?

Thanks.

Jerry8538, 2020-09-28:
@rizdam I am not sure if they will implement it, but for now, you could use a USB camera, which will work perfectly.

MadTooler, 2020-09-28:
Thanks for the reply.

I need to use the pi camera for my robots.   When a USB camera was properly recognized, it was quick.  Older USB cameras I tried were choppy.

I also found the sound/mic inputs were problematic without a GUI ability to change input volume or gain.

Jerry8538, 2020-09-29:
Okay, so maybe you could use another OS for it, as the camera works great on Raspbian, and it has all of the settings you need.
Unless you need NaCl support, for now, I think Raspbian is more stable and usable.
I am sure the devolopers will add it, but they have said that they will fix the performance issues first, so it might be a while before there is support for the Pi Camera.

yunshansimon, 2020-10-09:
The basic release only supports the USB camera. There is a lot of work to do to support the MIPI camera. We can't afford it without financial support.

closed 2020-12-13 by fydeos-alpha