FydeOS/chromium_os-raspberry_pi #92: Facing error in build_packages command for Raspberry Pi

akshaychoudhari358, 2020-09-08:
Hi Team,

I am trying to build Chromium OS for Raspberry Pi. However I am facing error in command "./build_packages --board=rpi3 --nowithautotest".

Error Message:
* ERROR: media-libs/mesa-9999::baseboard-rpi3 failed (depend phase):
 *   EAPI=7 is not supported
 *
 * Call stack:
 *               ebuild.sh, line 616:  Called source '/mnt/host/source/src/overlays/baseboard-rpi3/media-libs/mesa/mesa-9999.ebuild'
 *        mesa-9999.ebuild, line  16:  Called inherit 'base' 'multilib' 'flag-o-matic' 'meson' 'toolchain-funcs' 'git-2' 'cros-workon'
 *               ebuild.sh, line 317:  Called __qa_source '/mnt/host/source/src/third_party/portage-stable/eclass/meson.eclass'
 *               ebuild.sh, line 112:  Called source '/mnt/host/source/src/third_party/portage-stable/eclass/meson.eclass'
 *            meson.eclass, line  39:  Called die
 * The specific snippet of code:
 *       *) die "EAPI=${EAPI} is not supported" ;;
 *
 * CWD: /usr/lib64/python3.6/site-packages
 * S:   /build/rpi3/tmp/portage/media-libs/mesa-9999/work/mesa-9999
Error processing media-libs/mesa-9999, continuing...

Requesting you to provide any suggestions on the same.
[image]

yunshansimon, 2020-09-08:
1. Updating your chromium's overlays to r83.
2. Stopping work on "mesa", by taping "cros_workon-rpi3 stop mesa"
3. Building packages again or building a single package by the command
"emerge-rpi3 mesa". The current version of mesa is "20.0".

akshaychoudhari358, 2020-09-08:
Hi Team,

Thank you for your quick response.

I am trying to build Rpi using r80, I have taken checkout of chromium's overlays r80.
I am facing 'widevine_gni.patch' error while building using '/build_packages --board=rpi3 --nowithautotest' command.
Please provide your input on the same. Kindly refer screenshot for more details.
[image]

Also I wanted to know that which checkout is better r80 or r83.
Please provide you valuable feedback. 

Regards,
Akshay Choudhari

yunshansimon, 2020-09-09:
"CHROME_ORIGIN" is set to "LOCAL_SOURCE", the source already includes the
patch. Unset "CHROME_ORIGIN" or edit "project-fyde-for-rpi
<https://github.com/FydeOS/chromium_os-raspberry_pi/tree/r80/project-fyde-for-rpi>
/chromeos-base
<https://github.com/FydeOS/chromium_os-raspberry_pi/tree/r80/project-fyde-for-rpi/chromeos-base>
/chromeos-chrome/chromeos-chrome-80.0.3987.89_rc-r1.ebuild
<https://github.com/FydeOS/chromium_os-raspberry_pi/blob/r80/project-fyde-for-rpi/chromeos-base/chromeos-chrome/chromeos-chrome-80.0.3987.89_rc-r1.ebuild>",
remove "use widevine && patch_widevine".

akshaychoudhari358, 2020-09-10:
Hi, 

Thank you for you continuous support.
I am getting error on build packages command.

Below is the error:

chromeos-chrome-80.0.3987.89_rc-r1: clang-10: error: linker command failed due to signal (use -v to see invocation)
chromeos-chrome-80.0.3987.89_rc-r1: ninja: build stopped: subcommand failed.
chromeos-chrome-80.0.3987.89_rc-r1:  * ERROR: chromeos-base/chromeos-chrome-80.0.3987.89_rc-r1::fyde-for-rpi failed (compile phase):
chromeos-chrome-80.0.3987.89_rc-r1:  *   (no error message)

Please find attached screenshot for the same. Requesting you to provide feedback.
[image]

Regards,
Akshay Choudhari

yunshansimon, 2020-09-10:
Hi, Akshay,
  It said your chrome source doesn't exist at
'/home/ubuntu64bit/chrome_root'. Pls, ref to simple_chrome_workflow
<https://chromium.googlesource.com/chromiumos/docs/+/master/simple_chrome_workflow.md>
 .
  You can download a basic chrome source from chromium for Rpi
<https://github.com/FydeOS/chromium-raspberry_pi> to
'/home/ubuntu64bit/chrome_root/src'.
  I think the best solution is removing 'CHROME_ORIGIN=LOCAL_SOURCE' from
'overlay-rpi3/make.conf'.
Regards,
Yang Tsao

akshaychoudhari358, 2020-09-15:
Hi Team,

I am trying to build r83 branch for RPi3. However I am facing issue in build packages command.

Please find error messages below:
emerge: there are no ebuilds to satisfy "~chromeos-base/chrome-icu-83.0.4103.44_rc" for /build/rpi3/.
(dependency required by "chromeos-base/chromeos-chrome-83.0.4103.44_rc-r1::fyde-for-rpi" [ebuild])
(dependency required by "virtual/chromeos-interface-1-r5::chromiumos[-chromeless_tty]" [ebuild])
(dependency required by "virtual/target-chromium-os-1-r136::chromiumos" [ebuild])
(dependency required by "virtual/target-os-1-r3::chromiumos" [ebuild])
(dependency required by "virtual/target-os" [argument])
23:28:44: ERROR : emerge detected broken ebuilds. See error message above.

Kindly refer attached screenshot for more details.
Requesting you to provide feedback on the same.
[image]

fydeos-alpha, 2020-12-16:
dup #95

closed 2020-12-16 by fydeos-alpha