FydeOS/chromium_os-raspberry_pi #93: build error: failing for gclient sync command

sujaydesawale, 2020-09-08:
Hi,

I am using r 83 branch for RPi build. I am getting the error for gclient sync command which is attached in the .txt file.
Please have a look at the screenshot.

Regards,
Sujay
[r83_branch_gclient_sync_issue.txt](https://github.com/FydeOS/chromium_os-raspberry_pi/files/5186525/r83_branch_gclient_sync_issue.txt)
[image]

yunshansimon, 2020-09-08:
update your deploy tools and try again.

sujaydesawale, 2020-09-08:
Hi Yunshansimon,

Could you please tell me how to update deploy tool? 

Regards,
Sujay

yunshansimon, 2020-09-09:
Hi, Sujay,
  Pls, ref to install/update deploy tools
<http://commondatastorage.googleapis.com/chrome-infra-docs/flat/depot_tools/docs/html/depot_tools_tutorial.html#_setting_up>
Regards,
Yang.

closed 2020-12-13 by fydeos-alpha