FydeOS/chromium_os-raspberry_pi #95: [Issue report]

akshaychoudhari358, 2020-09-13:
Hi Team,

I am trying to build r83 branch for RPi3. However I am facing issue in build packages command.

Please find error messages below:
emerge: there are no ebuilds to satisfy "~chromeos-base/chrome-icu-83.0.4103.44_rc" for /build/rpi3/.
(dependency required by "chromeos-base/chromeos-chrome-83.0.4103.44_rc-r1::fyde-for-rpi" [ebuild])
(dependency required by "virtual/chromeos-interface-1-r5::chromiumos[-chromeless_tty]" [ebuild])
(dependency required by "virtual/target-chromium-os-1-r136::chromiumos" [ebuild])
(dependency required by "virtual/target-os-1-r3::chromiumos" [ebuild])
(dependency required by "virtual/target-os" [argument])
23:28:44: ERROR   : emerge detected broken ebuilds. See error message above.

Kindly refer attached screenshot for more details.
Requesting you to provide feedback on the same.

[image]

Thanks and Regards,
Akshay Choudhari

vikaspac, 2020-09-27:
+ me

vikaspac, 2020-09-27:
++ me also

yunshansimon, 2020-10-09:
Checking your Chromium OS source code is based on r83. The full branch name: "release-R83-13020.B".

PatrickChenHZ, 2020-10-14:
I have the same issue today, and I came to this post, but it's for sure my branch mathches, but it still throw out exact same error. I am on R83

nduijvelshoff, 2020-11-27:
@yunshansimon any update on this issue?

```
Scanning Configuration files...
Exiting: Nothing left to do; exiting. :)
13:23:07 INFO    : Running 'eclean -d packages' to clean up stale binpkgs
 * Building file list for packages cleaning...
 * Your packages directory was already clean.
13:23:07 INFO    : Elapsed time (update_chroot): 0m39s
13:23:07: INFO: Creating sysroot directories.
13:23:07: INFO: Installing configurations into sysroot.
13:23:07: INFO: Setting up portage in the sysroot.
13:23:08: INFO: Selecting profile: /mnt/host/source/src/overlays/overlay-rpi3/profiles/base for /build/rpi3
13:23:08: INFO: Updating toolchain.
13:23:08: INFO: Cross-compiler already up to date. Nothing to do.
13:23:08 INFO    : Clearing shadow utils lockfiles under /build/rpi3
13:23:09 INFO    : PORTAGE_BINHOST:  gs://chromeos-prebuilt/board/arm-generic/postsubmit-R86-13420.0.0-36875-8871426066622112112/packages
13:23:09 INFO    : Rebuilding Portage cache
13:23:11 INFO    : Checking package dependencies are correct: virtual/target-os virtual/target-os-dev virtual/target-os-factory virtual/target-os-factory-shim virtual/target-os-test
!!! CONFIG_PROTECT is empty for '/build/rpi3/'

These are the packages that would be merged, in order:

Calculating dependencies  . .. ... done!

emerge: there are no ebuilds to satisfy "~chromeos-base/chrome-icu-86.0.4240.28_rc" for /build/rpi3/.
(dependency required by "chromeos-base/chromeos-chrome-86.0.4240.28_rc-r1::fyde-for-rpi" [ebuild])
(dependency required by "virtual/chromeos-interface-1-r5::chromiumos[-chromeless_tty]" [ebuild])
(dependency required by "virtual/target-chromium-os-1-r143::chromiumos" [ebuild])
(dependency required by "virtual/target-os-1-r3::chromiumos" [ebuild])
(dependency required by "virtual/target-os" [argument])
13:23:17 ERROR   : emerge detected broken ebuilds. See error message above.
(cr) (release-R86-13421.B/(7bd3fbb...)) nick@chromium-os-build-machine ~/trunk/src/scripts $ 
```

PatrickChenHZ, 2020-11-27:
> @yunshansimon any update on this issue?
>
> ```
> Scanning Configuration files...
> Exiting: Nothing left to do; exiting. :)
> 13:23:07 INFO    : Running 'eclean -d packages' to clean up stale binpkgs
>  * [...]

I also failed to figure this out. I had tried everything recently. New discovery is that we have to use their chromium source, and must have it locally setup, you can refer to build.conf file to find this. But The error didn't change with local chromium source. I also tried following Emerge official doc to trouble shoot, and run emerge-rpi4 chromeos-base/chrome-icu    This end up with another error complaning about cannot find package NSS. And Even I installed NSS and PKG_CONFIG manually in chroot, it still won't be able to find it. Another Attmpt I made is compiling chromium out side chroot, and hope it will also compile chrome-icu, but it end up not compiling and throw error"undefined symbol: chromeos::DBusClientFydeOS::DBusClientsFydeOS(bool)"

Considering they are still updating this repo, I think the instruction is just too old, or they add something that need extra steps.

fydeos-alpha, 2020-12-16:
A refactoring PR has been in place since today, there are quite a few changes in terms of how we organise dependecies and required packages. The build workflow was tested working, please retry.

To avoid any unnecessary issues caused by your legacy code, it's recommended if you could begin from scratch.

closed 2020-12-16 by fydeos-alpha