FydeOS/chromium_os-raspberry_pi #96: [Issue report] Lag on Raspberry Pi 3B+

LunatiikXD, 2020-09-22:
**Describe the bug or issue**
Installing ChromiumOS 83 lags

**Describe your setup**
 - Raspberry Pi 3 B+
 - ChromiumOS 83
 

**To Reproduce**
Steps to reproduce the behavior:
1. Install and boot the Pi.
2. The lag will start in Setup

**Expected behaviour**
A lag on the Pi.

Jerry8538, 2020-09-28:
Maybe the Pi 3 B+ just isn't capable of running it decently?
How much RAM do you have @LunatiikXD ?
If it is less than 2 gigs, you are unlikely to have good performance.

LunatiikXD, 2020-09-28:
1 gb of ram. However 77 and 80 runs fine

Jerry8538, 2020-09-28:
Oh, so it's just this version then. I am not too sure what's going on, but if the previous version works for you, why do you wanna upgrade to this version?

LunatiikXD, 2020-09-28:
For the Zoom support.

Jerry8538, 2020-09-28:
Oh yeah, well, I am stumped as to what could be causing this apart from just some rogue app or extension, but I think it's probably just an optimization thing. And I obviously don't know if or when the developers will fix it but they will be aware of this issue, so hopefully we can see a fix soon.
Sorry for not being very useful though

yunshansimon, 2020-10-09:
Pls, remove most of the extensions running in the background.

LunatiikXD, 2020-10-09:
How do i do it? Even in the setup it lags.

fydeos-alpha, 2020-12-12:
please try r86r1 release, tested with Pi 3B 2GB RAM, I would say it's acceptable performance.

closed 2020-12-12 by fydeos-alpha