FydeOS/chromium_os-raspberry_pi #97: [Issue report]

JTthePres, 2020-09-27:
I can't use hdmi audio, i tried to follow issue #17 istruction but i can't undesrtand if i've made an error or if it's a bug.
What i must modify at file /etc/init/force_audio_output_to_headphones.conf ? I've just changed the last number inside the script from 2 to 0 but it still doesn't work .
[image]

I've just using a raspberry pi4 model b with the image r83r1 plugged to a sony monitor. 
Someone can help me?
Thank you

Jerry8538, 2020-09-28:
I am having the exact same issue. I am on the same device, and I have tried setting it to 0, 1 and 2 with no effect.
Thanks in advance!

FiodorTretyakov, 2020-10-06:
I have the Samsung TV and the problem exactly the same - it not allow me to select the audio device output at the system  anymore - but no audio at all with 0-setting - no hdmi, no headphones.

xyploid, 2020-11-07:
hi All, i am pretty new to this. can you let me know what is the key board shortcut to save the edited file.

Jerry8538, 2020-11-07:
@xyploid 
Press Shift + Z + Z.
Yes, that's right, you gotta press Z twice.

yunshansimon, 2020-11-25:
The script was updated many times, so the document lost the track. Delete force_audio_output_to_headphones.conf or rename it to force_audio_output_to_headphones.conf.bak will fix the issue.

closed 2020-12-13 by fydeos-alpha

adsports, 2020-12-13:
> @xyploid
> Press Shift + Z + Z.
> Yes, that's right, you gotta press Z twice.

not work
I try to enter with sudo but it tells me, ¨we trust have ..... ¨ and asks for my password.

How can I give write permission?

Fusion3Design, 2020-12-14:
this does not appear working and the latest release the force_audio.....file is not in /ect/init any longer

adsports, 2020-12-15:
> this does not appear working and the latest release the force_audio.....file is not in /ect/init any longer

This version yes r83, but notpermision for edit the file