FydeOS/chromium_os-raspberry_pi #98: Video extremely laggy/buggy and sometimes the screen goes black

Botspot, 2020-09-29:
Several people I know have also experienced this with the more recent releases of FydeOS.
On a RPi4, the video performance is very weird.

For example, when scrolling down a page, sometimes it jumps back up the page, for a split second. It's like CrOS is remembering past frames of the animation and playing them back in the wrong order.
This frame-jumbling is also noticeable while playing videos and doing a video call.
It also occurs when editing in-browser 3D files within TinkerCAD. Trying to pan around the model, it jumps around so badly, that I didn't know which way the model was facing, because it kept showing prior frames.

Additionally, the screen goes black randomly. Others have noticed this too, so this isn't an issue with my Pi, my hdmi cable, or my screen.

Today, I noticed that the mouse pointer randomly glitches into a pink polkadot square about 48x48 pixels in dimensions.

What is happening, and how to fix it?

Jerry8538, 2020-09-29:
I have experienced all of these except the polkadots, and it is honestly very strange.
I have also seen that when I play audio, on Zoom or on a Youtube video, it just randomly cuts off, and I have to reselect the audio device in the sound settings.

yunshansimon, 2020-10-09:
We have fixed the issue at the latest commit. Stay tuned for the next release.

closed 2020-12-12 by fydeos-alpha