FydeOS/chromium_os-raspberry_pi #99: Running Linux program Android Application

auto-p, 2020-10-01:
How to run an Android application and run a Linux application? After I finished the installation, I logged into the Google account system and did not automatically install anything except plug-ins I have in Google,

Jerry8538, 2020-10-01:
You can't install Android or Linux apps afaik. This is just regular Chromium, and right now there aren't any plans for Android app support.

yunshansimon, 2020-10-09:
Maybe we can provide crostini in the future for rpi4. But I don't think it is an attractive idea for rpi4.

closed 2020-12-12 by fydeos-alpha

fydeos-alpha, 2021-01-18:
@au-top  @Jerry8538   https://releases.fydeos.io/11.4/rpi4-fydeos you may want to check out this one, it's FydeOS for You, not Chromium OS, and it supports Android and Linux.

Jerry8538, 2021-01-19:
@fydeos-alpha That's amazing! I have a few questions, however. How exactly do I use Android apps on FydeOS? It doesn't mention that anywhere, and what exactly does it support as of now, like does the GPU work, and WiFi etc. If everything works correctly, I might just use FydeOS, because running Android apps is pretty sweet!

fydeos-alpha, 2021-01-19:
@Jerry8538  I believe [this video](https://www.youtube.com/watch?v=ny38ubq22hQ) has got all your questions explained, and perhaps [this one too](https://www.youtube.com/watch?v=U4i33CSPGnM).

au-top, 2021-01-19:
use balenaEtcher out data a fail ,  I tested it many times , Is there a problem like mine ?

au-top, 2021-01-19:
I checked that sha256sum is correct

fydeos-alpha, 2021-01-19:
run etcher as Administrator if you are using Windows. Otherwise, try
Raspberry Pi Imager

au-top, 2021-01-19:
The Mac OS I borrowed was successfully written, but after I inserted the raspberry pie and connected it to the HDMI monitor, I successfully displayed the rainbow test picture of raspberry pie when it was turned on, and then there was no picture. Later, only the monitor indicated that there was no signal. The hard disk light (green light) of my raspberry pie flickered from time to time. I tried to restart the power after 40 minutes, but the problem remained, My raspberry pie 4b4g

fydeos-alpha, 2021-01-19:
Seems the kms graphics mode isn't compatible with your monitor. Try editing
the config.txt and change it to fake-kms mode to achieve some extra
compatibility:

https://github.com/FydeOS/chromium_os-raspberry_pi/issues/16#issuecomment-586219347

Jerry8538, 2021-01-23:
> @Jerry8538 I believe [this video](https://www.youtube.com/watch?v=ny38ubq22hQ) has got all your questions explained, and perhaps [this one too](https://www.youtube.com/watch?v=U4i33CSPGnM).

Ah, my bad for not looking it up first, but thanks a lot for linking them!

rimonmath, 2021-10-18:
> Maybe we can provide crostini in the future for rpi4. But I don't think it is an attractive idea for rpi4.

Please provide. I need it