FydeOS/chromium_os-raspberry_pi #100: Security Concern

Coda-Coda, 2020-10-11:
I see in the releases it states:
> This release is of a test build, which is by design very insecure (for any production environment), as quoted from Viper: "You must never log in to them with credentials you care about. It will automatically start ssh with root access via well-known credentials."

If I am security conscious, should I not use this build at all except for testing purposes? If I were to disable ssh and/or change the root password would that resolve this security issues?

I would like to be able to use this for everyday computing running Chromium OS, but this concern is holding me back. Please can you clarify what it would take for this image to be installed and used in a reasonably secure manner?

fydeos-alpha, 2020-10-11:
Hi, the goal for this project was never intended to provide a production-grade image that can be used in a production environment or at any commercial capacity. We release "test" pre-built images to encourage more users to try Chromium OS on Raspberry Pi devices.

I am not sure how you define "security conscious" and "reasonable secure manner" - these descriptions do vary quite a lot depending on the context. In any way, you should not deploy the "test" image onto any production environment. Disabling ssh / changing root password may solve some security concerns, but the overall security of these Chromium OS builds are still far less than what Chrome OS could offer.

Coda-Coda, 2020-10-11:
Thanks for the helpful reply. You are right in saying that "security conscious" and "reasonably secure manner" are vague terms. I am just really wondering whether installing Chromium OS via this build will *definitely* leave an obvious security hole and if so what steps I can take to mitigate that (is just changing the root password sufficient for that?). I understand there may be generic issues with Chromium OS vs Chrome OS, but in any case I am not aware of any way to install Chrome OS on my Raspberry Pi.

To be clear, this isn't for any commercial project, just for personal use. But even for personal use I'd like to know that, at least, the obvious security holes such as leaving ssh access open with widely known passwords are resolved (or resolvable).

Especially with guides like [this one](https://www.iottechtrends.com/install-chromium-os-on-raspberry-pi/) pointing users here, I feel it would be a good idea to make it clear how to minimise the security issues. Alternatively, tips on how to build Chromium OS (or Chrome OS) in a more secure manner would be awesome.

Thanks for your great work on this project 😀

Coda-Coda, 2020-10-11:
Much as I would have liked to try use Chromium OS, because of these issues I’ll use Ubuntu Mate instead. I realise that creating an image for general purpose use isn’t the intent of this project.

All the best with it though!

closed 2020-12-12 by fydeos-alpha