FydeOS/chromium_os-raspberry_pi #101: Can't get past "searching for bluetooth and usb devices" screen

vladimirgluten, 2020-11-05:
I'm on the screen, it finds my mouse and KB, but clicking on Continue does nothing.  it's registering that I click, because I can see the little animation.  The Shutdown button does work, however.  This is 83.0.4103.45 (Platform 13020.39.2020_7_12_1601 (test build - yang) developer-build rpi3)  Just downloaded it today.

yunshansimon, 2020-11-25:
I have no idea what's you're talking about. Can you take a screenshot?

vladimirgluten, 2020-11-25:
I took a pic with my phone, since I can't take a screenshot with ChromeOS, partly because I'm not loaded fully in and partly because I don't even know how to do a screenshot on ChromeOS (never actually used it before)
[image]

When I click on the Continue button, nothing happens, it doesn't continue to load the OS

yunshansimon, 2020-12-02:
We will release a new version with more features and more stability, please try the new one.

closed 2020-12-13 by fydeos-alpha