FydeOS/chromium_os-raspberry_pi #102: Need help with widevine

Botspot, 2020-11-06:
Note: This is not a question pertaining to FydeOS! But I know you guy sare well-versed in chromium-related things, and (hopefully) will have an idea for how to fix this issue:

Raspbian just upgraded to Chromium v84. Chromium v84 no longer works with Chromium Media Edition, so many Pi users are not able to watch Netflix on their Pies anymore until it works again.  
I have [provided an easy way](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=290329) to revert back to Chromium v78, but (obviously) that's not ideal for long-term usage.  

I'm hoping somebody in this repo will see this and be willing to lend a hand. Here's the main RPi Forum thread about it to start you off: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=290239

Botspot, 2020-11-06:
Never mind, I figured it out.

Jerry8538, 2020-11-07:
How did you go about fixing it @Botspot ?
Could you please tell us what you did to figure it out?

Botspot, 2020-11-07:
> How did you go about fixing it @Botspot ?
> Could you please tell us what you did to figure it out?

Sure. I had been monitoring the "Today's latest Chromium update (v84) breaks Netflix playback" RPi forum thread, and within a few hours, [somebody said](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=290239&p=1755945#p1755154) they found a script which fixed it for them.  
Next step was finding the exact same script and replicating their results. That took longer, but [ShiftPlusOne set me on the right track](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=290239&p=1755945#p1755836).  
Turns out that both of these files are necessary: `/opt/WidevineCdm/_platform_specific/linux_arm/libwidevinecdm.so` and `/opt/WidevineCdm/manifest.json`.
Next step was to upload the files in an efficient way. (as the existing `widevine-flash_armhf.sh` downloaded a 2GB file) [Here](https://github.com/Botspot/chromium-v84-widevine)'s my repository.  
And lastly, I added it to Pi-Apps in order to make a custom Menu launcher to work. You can read my announcement [here](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=290443).

Botspot, 2020-11-11:
> Thanks!

@Jerry8538 
@fydeos-alpha 

There, I helped you, so would you mind helping me? Raspbian's Chromium does not include NaCl support, and that's why Zoom, VNC, Teamviewer, and most other Chrome Apps don't work. [This RPi engineer](https://github.com/RPi-Distro/chromium-browser/issues/7) is considering trying to add it, but needs some guidance. Would either of you be willing to help him?

Botspot, 2020-11-14:
Anything?

Jerry8538, 2020-11-16:
@Botspot I replied to you in your discord server.
I said that I don't knw of a way except just building it with the nacl flag set to true, which you probably have tried already, so I don't think I can be of much help, but if someone else has any idea as to how we can get nacl, it would be greatly appreciated.

yunshansimon, 2020-11-25:
Open an issue for the problem with zero relationship with us, are you serious?

Botspot, 2020-11-25:
> Open an issue for the problem with zero relationship with us, are you serious?

I thought I would try.  
It's because you guys added NaCl to FydeOS, so I thought that maybe you guys would be familiar with the steps involved to make it work on Raspbian.

Now, as @Jerry8538 says he can't help, then I guess I can close the issue.

closed 2020-11-25 by Botspot