FydeOS/chromium_os-raspberry_pi #103: Wont boot

anaskatt01, 2020-11-11:
My raspberry pi 3b+ won't boot

yunshansimon, 2020-11-25:
ChromiumOS for Rpi doesn't support boot from usb. I don't know how to boot from usb either. Maybe we can take a shot with the latest release. You can modify the `cmdline.txt` in partition 12.  Change `root=/dev/mmcblk0p3` to `root=/dev/sda3` . Tell me the result wether or not it works.

anaskatt01, 2020-11-26:
> ChromiumOS for Rpi doesn't support boot from usb. I don't know how to boot from usb either. Maybe we can take a shot with the latest release. You can modify the `cmdline.txt` in partition 12. Change `root=/dev/mmcblk0p3` to `root=/dev/sda3` . Tell me the result wether or not it works. So thankyou for the response however i am no longer looking into this project because of some technical issues so please close this question

closed 2020-11-26 by anaskatt01