FydeOS/chromium_os-raspberry_pi #106: Would it be possible to chroot into FydeOS from Raspbian?

Botspot, 2020-11-24:
The title says most of it.

I've developed and currently maintain vdesktop - a virtual machine software to run RPi OS'es on Raspberry Pi.  
Chromium OS support would be very cool.

How would that work? I'm not familiar with how chromium OS works under the hood. Does that type of linux have a root filesystem somewhere, containing folders like /dev, /sys, /bin, and /usr?

yunshansimon, 2020-11-25:
The chromium for ChromiumOS needs more kernel modules which don't include in RpiOS kernel. You need to combine two kernel config files, some of the flags conflict each other. I don't think it will work.

Botspot, 2020-11-25:
@yunshansimon Would it be possible to access CrOS's linux subsystem? And not Chromium?

yunshansimon, 2020-12-02:
We provide "crostini" in the next release. It's a virtual machine running Debian Linux.

closed 2020-12-12 by fydeos-alpha